sqfmi/badgy #3: GxEPD repo does not package ZIP files for use of Hello example

sanecito, 2018-08-06:
In the badgy example it says to include the library from https://github.com/ZinggJM/GxEPD. All the other libraries we need to include have zip bundled releases, but GxEPD doesn't How do we install GxEPD w/o using a .zip file?

Can we use the Adafruit EPD by Adafruit (v1.0.1) library that's findable via the Library Manager instead?

Ryomoo, 2018-08-06:
Download the master branch as .zip and install it in adruino.

sqfmi, 2018-08-08:
Hi @sanecito is this resolved? If so, we'd like to close this issue.

closed 2018-08-09 by sqfmi

pastorhudson, 2018-10-13:
[Here's the instructions to get the correct branch: https://github.com/sqfmi/badgy/issues/18#issuecomment-429561161](https://github.com/sqfmi/badgy/issues/18#issuecomment-429561161)