sqfmi/badgy #6: Battery voltage

gordonendersby, 2018-08-18:
Is it possible to measure the battery voltage?

Or have you considered using the adc with a voltage divider?

sqfmi, 2018-08-22:
Hi @gordonendersby!

Currently the ADC is not in use for Badgy, so you can connect a voltage divider (e.g. 220K/100K) and connect it to the ADC pin. This may be available in the next version :). Thanks!

gordonendersby, 2018-08-25:
Ive worked this out now after reminding myself how to measure the internal vcc.

The internal ADC can measure the battery. No need for voltage dividers unless you want to measure an outside source or other analogue source.
Ideally before the wireless is turned on due to drain.

`/* Needed to point adc at internal vcc*/
ADC_MODE(ADC_VCC);

/* in startup */
float vccVolt = ((float)ESP.getVcc())/1024;
`
It would be great if you just broke out any spare pins to pads that a pin header can be connected too.

gordonendersby, 2018-08-25:
Closed as you can use internal ADC to VCC

closed 2018-08-25 by gordonendersby

sqfmi, 2018-08-25:
The internal ADC measures the input voltage to the ESP, which is regulated by the LDO, not the actual battery's voltage.