sqfmi/badgy #7: 3d CAD files for case printing?

lasniko, 2018-08-21:
I wanted to put my badgy in a nice 3d printed casing. Does anyone have a fitting 3d model?

chunkysteveo, 2018-08-21:
Would also like this when it arrives - was waiting on the board files to import into Fusion to then work from that. If you can wait till I get mine, and we get the board schematics - I can create one for you?

sqfmi, 2018-08-27:
We're trying to export the STEP file for the board but we're still missing some 3D models (e.g. FPC connector, battery clip, etc.) Once we have those created (or if someone can point us to where we can find them) we'll put it on GitHub right away

chunkysteveo, 2018-08-28:
Have you checked "GrabCAD"? It's a great user resource of 3D objects.

Pablo2048, 2018-08-29:
You can also try 3dcontentcentral.com , and/or http://www.pcb-3d.com/electric_type/ffc-fpc-flat-flexible-connectors/ ...

chunkysteveo, 2018-08-29:
This is probably what you need for your 5-way - https://grabcad.com/library/thmu-13-smd-joystick-it-1501-ts-1600e-1. Looks just the part?

chunkysteveo, 2018-08-29:
Actually fits perfect according to the holes in the PCB STEP file! Do you need the e-ink display creating, as that is missing?

sqfmi, 2018-08-29:
Yes and thank you for submitting the PR! The 3D model is now complete! 🍾

chunkysteveo, 2018-08-30:
Eeeek... now I should think about a cool case!?!

sanecito, 2018-09-07:
Nice! I had made one out of Legos as a placeholder case. If anyone designs a case for badgy (especially one that supports gluing magnets to the back) please link to it so I can 3D print it!

chunkysteveo, 2018-09-07:
I'm in the middle of designing some cases... all theoretical as I don't have the Badgy yet. So I guess the idea is to stick it to a fridge or something? Sounds cool! If I get time, I think I will make a few case versions (I just like to mess about in Fusion360!!) so a magnet version that sits flush is on the cards now! What size magnets, like the little 5/6mm circle NEO ones?

svenove, 2018-09-07:
> So I guess the idea is to stick it to a fridge or something?

I want to attach Badgy to my shirt, as a name badge. :) Magnets will probably be the best solution any way, with magnet on the inside of the shirt.

chunkysteveo, 2018-09-07:
Ah yes, magnets would work well too for clothing.

gwtzt, 2018-09-07:
Stephen,
I ordered one more than I need. I could send you one if you send me a matching case in return. 
Regards,
Stephan

chunkysteveo, 2018-09-07:
Ah that's very generous of you! I have one coming my way, but the definition of "having one more than I need" does not compute for me!! hahaha! I'd be interested in a service swap though and could send a case out for a swap once designed and printed. Whereabouts are you in the World?

sanecito, 2018-09-07:
I use an Adafruit magnet that's about 45.3mm x 13.3mm x 5.1mm: https://www.adafruit.com/product/1170
I know some 3D print designs use holes in say, a wall of a box or what have you to save on material, weight, time, etc. So long as there's at least a chunk of back surface for all the adhesive for the magnet that would be fine for how I would use the case.

chunkysteveo, 2018-09-07:
That's a big magnet! I will take the dimensions though and take that on board.

Steve

svenove, 2018-09-07:
> I use an Adafruit magnet that's about 45.3mm x 13.3mm x 5.1mm: https://www.adafruit.com/product/1170

Since they charge 17$ for shipping, I'd rather try something like these:
https://www.aliexpress.com/store/product/10pcs-45-13mm-Wholesale-Strong-Magnetic-Name-Tag-ID-Badge-Holder-with-3M-Adhesive-on/3011056_32811317834.html

But, as @sanecito says - as long as there is a flat surface on the back, anything/any size can be attached by adhesive. :)

chunkysteveo, 2018-09-07:
Yeah, that's way cheaper! Lets just say there will be a 46 x 14 mm indent in the back to locate a bar magnet :)

sanecito, 2019-03-06:
@chunkysteveo Have you finished any of your case designs by chance?

chunkysteveo, 2019-03-07:
Hi @sanecito - I did do a test case for the original Badgy, which can be found in the still open Pull Request on this repo, you can find files to download in the comment thread. I'm not 100% happy, and it seemed a little over engineered for the task. But have not done a follow up design, yet. I did then create a more simpler "bumper" style case that can be used as a badge/lanyard too, and I have made it compatible with both the old and new versions of Badgy - this can be found here - [badge-frame_all_v3.zip](https://github.com/sqfmi/badgy/files/2460886/badge-frame_all_v3.zip). Due to all the issues with the new Badgy serial and battery problems it's not been something I have been pushing myself to do. There are some designs on Thingiverse too, which I will eventually put mine on there too.

Steve

Xaldimo, 2019-05-18:
There is also a case on Thingiverse:

https://www.thingiverse.com/thing:3312442

chunkysteveo, 2019-05-22:
My bumper case(s) are now on Thingiverse https://www.thingiverse.com/thing:3647429