sqfmi/badgy #8: Can't upload to device

PlastiBots, 2018-08-21:
Hey guys..  Cool device here.  Looking to hack it into a weather unit.  I followed  the instructions to get it on my WiFi and that works.  However, now I can't seem to access the device or ping it.  When starting I hold the center button and power on. Then get to the "_you are now connected....   go to 192.168.2.44:8888 to upload a sketch_". However, when I go to that IP in Chrome I get nothing - cant find it.  Ping also does not work.  The device shows on my router as connected and has a strong wifi signal.  Any ideas?  Note that I am powering via the PC USB cable.  Also no port shows up in Arduino IDE.  Am I correct that this does not support upload via USB as I see no USB/TTL converter on board?

Details:  Win 10.  Arduino 1.8.5.  All libraries installed and compiling fine.  Choose Generic ESP8266 module.   Followed instructions here: https://github.com/sqfmi/badgy

Q: When compiling, what board should be selected?

sqfmi, 2018-08-22:
Hi @PlastiBots!

Do you know if your computer and Badgy are both on the same network/subnet? If they are you should be able to connect and upload over OTA. Check for any settings on your computer/router that may be block port 8888. You can choose the board "NodeMCU 1.0 (ESP-12E module)" with all other settings at default for compiling.

Let us know if this works, thanks!

PlastiBots, 2018-08-22:
Hey.  I figured it out.   I had to enable UPnP on my router.  Once I did that, I was able to access the page.

sqfmi, 2018-08-22:
Awesome! Thanks for letting us know the solution, we'll add it to our troubleshooting guide.

PlastiBots, 2018-08-22:
Hey guys.. Spoke too soon.  Oddly, tried again today and it's failing again - even with UPnP on.  I think it's an issue with my router/UPnP as it definitely worked when I first turned it on.  I've tried turning it off, then back on but it still fails.  Will keep plugging away and report back.

vidax, 2018-09-19:
Okay, it is a badgy issue. Am working with Epeejunior, and his Badgy does not want to connect, so i pulled another badgy from the stack and it connects successfully to the same network. We even tried to renew the lease of the badgy on the router and the badgy showed a new IP, but still, no way to connect. So I do not think it is a network issue, but really a bug. 
I will try to connect this badgy at home (It will ask me to setup a new wifi). if it does not work, I will flash it via FTDI and report here.

sqfmi, 2018-09-20:
Thanks @vidax, please let us know the results.

vidax, 2018-09-20:
I turned on the badgy on another network. This time, the badgy does not refresh the screen and is stuck on telling me to login on the IP address of the other network. 
--> If I go back on the old network, the screen will refresh, show the same screen, or a new IP if we canceled the lease, but still no way to connect.

sqfmi, 2018-09-20:
When trying on another network, try holding the center button for at least 15 seconds after powering on. The WiFi manager will try a few times with the saved credentials, before it times out and starts the Badgy AP again to ask for new credentials.

Can you also confirm that the IP/Ports are not blocked on your PC/Router/Network? Try hosting a WiFi hotspot with your phone and see if you are able to connect. It does sounds like a networking issue so far, please do let us know your testing results so we can try to support, thanks!

sqfmi, 2018-10-06:
Closing. Please re-open if issue persists.

closed 2018-10-06 by sqfmi