sqfmi/badgy #9: Wake from DeepSleep with button?

svenove, 2018-08-24:
Is there any way to wake Badgy up by using the switch, when using ESP.deepSleep()?
Like if I want the HourlyQuote-example to also update when I press the button. :)

sqfmi, 2018-08-25:
Hi @svenove! Unfortunately no, the ESP8266 does not support GPIO interrupts to wake from deepsleep. The only way to wake the ESP8266 is to perform a power cycle (e.g. slide the power switch up and down) or by pulling the reset pin low.

svenove, 2018-08-25:
@sqfmi That’s what I thought, hmm...
What battery life can be expected when always awake? Like in the Spotify-example. I would expect that not using deepsleep has a huge impact on battery life?

gordonendersby, 2018-08-25:
Would be great if you could break out the reset pin or have pads that a momentary switch could be soldered on to.

In the meantime you could add your own push switch to the reset pin with a bit of careful soldering.
and mount the switch on whatever you are housing the badgy in.

svenove, 2018-08-26:
I guess calling wifi.stop() would reduce power consumption? At least if not reconnecting too often (as reconnect probably uses a bit more than keeping connected?). 

So instead of deep sleep, I could disconnect from WiFi instead?

Any other tips for reducing power consumption? :)

sqfmi, 2018-08-26:
Instead of deep sleep, you could also try "light sleep" mode which still allows you to wake on interrupt, and consumes ~1mA  (according to documentation. Check out this thread for some code snippets and details: https://github.com/esp8266/Arduino/issues/1381

closed 2019-05-18 by sqfmi