sqfmi/badgy #11: images on badgy

ac8vw, 2018-09-07:
I cannot find any documentation on the specs for formatting images for badgy other than using the xbm format. Things that I tried won't compile. Badgy is great; but, I'd like it to display more than "Hello my names is Badgy." I believe it would be a good service if some more documentation were available on hacking the files.

svenove, 2018-09-07:
I simply used an online “image to C array”-converter and replaced the variable-value in “hello.h”. Worked great. :)

Not sure if that’s what you’re asking, though..?

sqfmi, 2018-09-08:
Check out http://javl.github.io/image2cpp/ , you can load up an image file and have it converted to an array, simply replace that in hello.h or create a new header file. Images work best when the dimensions are multiples of 8 e.g. 8x8, 24x24, 64x128, etc. We're currently in the process of updating documentation and will post a tutorial shortly, let us know if you have any questions in the meantime. Thanks!

ac8vw, 2018-09-08:
The link is very helpful. Looking forward to the tutorial.

sqfmi, 2018-10-06:
Instructions updated in the README, closing this issue. Please re-open if issue persists.

closed 2018-10-06 by sqfmi