sqfmi/badgy #17: Board will not go into OTA mode

patrickdijusto, 2018-09-19:
I got the board working just fine out of the box.  I downloaded all the files needed to run "Hello".  I compiled it and uploaded it just fine. 

But now the board will not return to OTA mode.  I shut it off, press and hold the center button, turn on the board, and nothing happens.  The edisplay still shows the "Hello my name is..." graphic.

I have tried this while powering the board via USB from my laptop, and via USB from a fully charged power brick.  When I flip the board's switch to ON, a red LED at the center of the lower edge of the reverse side of the board blinks about once every 7 seconds, in sync with the blue LED on the ESP32.  That's all that happens.

When I flip the switch to OFF, the red LED at the center of the lower edge of the reverse side of the board blinks about 5 times per second. THe blue light does not blink at all.

I have tried multiple USB cables and power sources.  Nothing works to put the board into OTA mode.

I do not have the skills to solder connectors to the board. What can I do?

patrickdijusto, 2018-09-20:
The screen does NOT refresh when I flip the switch from on to off or off to on.  

I used the hello program from this git repo.  The only thing I changed in the hello program was my name.

sqfmi, 2018-09-20:
Thanks for the info. Could you share with us the compile settings used in Arduino? The blue LED on the ESP blinking every 7 seconds means it is crashing and resetting. Since the name is the only change to the code you made, it is likely to be either a compile settings error, or a hardware related issue (e.g. insufficient USB power, loose display connector, shorted components, etc.)

To flash your Badgy with a USB-Serial adapter, follow the wiring information in our [documentation](https://github.com/sqfmi/badgy/tree/master/examples) and push the pin headers against the board like so (you may have to scrape the surface a little for better contact):

[image]

sqfmi, 2018-10-06:
Closing. Please re-open if issue persists.

closed 2018-10-06 by sqfmi