sqfmi/badgy #18: Is there a compile problem with v3.0.0 of GxEPD Lib

seigel, 2018-09-25:
I have not exhausted all attempts yet, however I am getting a failure to build with weather.ino example.  It seems to be complaining most about the GxEPD lib.  Any chance I am missing something?

sqfmi, 2018-09-25:
Hi @seigel,

We've only validated with v2.x since v3.0.0 just got released last week. Please continue to use v2.x while we check out v3.0.0. Thanks!

seigel, 2018-09-25:
Will do, I'll see if I can get a copy and try it out :)

seigel, 2018-09-25:
Yup.  Checking out a previous rev (< 3.0.0) compiles.  Thanks!

chunkysteveo, 2018-09-25:
Ahaaa.... That makes sense now! Just started to play with the example scripts last night and didn't spot the v2.X requirement for the GxEPD libraries and was getting dozens and dozens of errors and warning when compiling. Adding in the latest V2 version which was only overwritten a few days ago has solved the compiling issues!

sqfmi, 2018-09-26:
Closing this issue. Please continue to use v2.x until we have validated and ready to update to v3.x

closed 2018-09-26 by sqfmi

pastorhudson, 2018-10-13:
Instructions for getting the correct version in case someone needs them:
This assumes you have git installed. If not [install git here](https://git-scm.com/book/en/v2/Getting-Started-Installing-Git) 

Once you have git installed:

Open terminal in mac & linux, or command line in windows.
Change to the directory where you want to put the library:
Windows is probably `cd %USERPROFILE%\Documents\Arduino\libraries`
Clone the GxEPD repository:
`git clone https://github.com/ZinggJM/GxEPD.git`
Change to the GxEPD folder:
`cd GxEPD`
Reset your local version to the version that works:
`git reset --hard 20eff1d80f7276a3c093183c3823cdd5f6cf1c9d`

You should now be on the branch that works. And your GxEPD library should be ready to go.