sqfmi/badgy #19: How to use it... as a badge

titimoby, 2018-09-25:
The project is amazing, I really like it.
But I don't see on pictures how it is supposed to be fixed.
Any holes usable?

chunkysteveo, 2018-09-25:
A 3D case would be ideal to have this wrapped in. With space for a magnet, and loop for a lanyard... watch this space! (but yeah, a hole in the PCB would have been super handy!!)

chunkysteveo, 2018-09-27:
[case_prototype_badgy_frame.zip](https://www.chunkymedia.co.uk/wp-content/uploads/2018/09/case_prototype_badgy_frame.zip)
Try this one @titimoby  - not printed yet, but if you have a 3D Printer, feel free to try it out.

titimoby, 2018-09-27:
I'll give it a try... as soon as I get my badgy (waiting for paycheck ;) )

richardvk, 2018-09-28:
I did a quick print this morning, turned out nice! I did need to cut a tiny bit away from the top left diagonal so i could squeeze the board in, but otherwise good!

[image]

chunkysteveo, 2018-09-28:
Wicked. And yes - I have just printed one out too and can't get it to "squeeze" into the last corner. I thought there would be enough wiggle room, but no. I will adjust the "top left" corner to be less to it will be small enough to push in, but still grip it a little. This frame can be the "flexible" version, as I am sure it will fit fine with some Ninja Flex or TPU type material... version 2 to do!

chunkysteveo, 2018-10-08:
Have made a new version of the Frame (frame) which should be easier to insert the Badgy to as it has a "lip" in the top left that should allow you to snap the Badgy into. Also made it compatible with both versions of the Badgy (Blue and White PCB versions) as the USB/switch actually switch over top to bottom, and the ESP chip has moved.

To insert, locate Badgy into the "right" two corners, then twist the frame into the bottom left corner, and the top left corner should snap in... printing tonight so not tested yet...

[badge-frame_all_v2.5.zip](https://github.com/sqfmi/badgy/files/2455392/badge-frame_all_v2.5.zip)

richardvk, 2018-10-08:
Thanks Stephen!

I will give this a go asap :)

chunkysteveo, 2018-10-09:
Ignore V2.5 above, try this version, especially if you have a new White PCB version. 
[badge-frame_all_v3.zip](https://github.com/sqfmi/badgy/files/2460886/badge-frame_all_v3.zip)

veggie2u, 2018-12-24:
@chunkysteveo are you able to upload a model file that can be edited? I wanted to make a version that had both tabs with the holes on the top.

chunkysteveo, 2018-12-24:
It's a Fusion360 project,  if that works for you? Or i could make the alterations for you if you want?

veggie2u, 2018-12-24:
Oh, cool. I am trying to learn 360 right now.

veggie2u, 2019-01-04:
@chunkysteveo don’t mean to pester, but are you able to post the 360 project?

chunkysteveo, 2019-01-04:
Hi @veggie2u - the 360 project is a mess, and not something I would like to publish, especially if you are learning it(!!) Took me about 30 seconds to make the attached STL file with clips on the top - let me know if that works for you? - Steve

[badge-frame_v3_top_top.zip](https://github.com/sqfmi/badgy/files/2726303/badge-frame_v3_top_top.zip)

chunkysteveo, 2019-01-04:
Apologies for the slow reply too - been busy with Christmas, New Year etc and not been around many computers.

veggie2u, 2019-01-04:
@chunkysteveo That's cool. I get it. So I did print it, and it looks like the switch opening got filled in. I went back and looked in PrusaControl and in Slic3r and sure enough, the stl doesn't have the opening.

chunkysteveo, 2019-01-04:
Ahh crud! Let me check the file!

chunkysteveo, 2019-01-04:
[badge-frame_v3_top_top_fix.zip](https://github.com/sqfmi/badgy/files/2728503/badge-frame_v3_top_top_fix.zip)

Sorry about that, here's the correct file with the updated gaps for the switch and USB! Sorry you had to print it before spotting it!

Steve

veggie2u, 2019-01-05:
It printed just fine. Thanks!

lasniko, 2020-01-10:
I recommend printing in some flexible filament, since it makes attaching the badgy much easier.
Here is the one I printed in nylon:
[badgy name tag case.zip](https://github.com/sqfmi/badgy/files/4044825/badgy.name.tag.case.zip)

chunkysteveo, 2020-01-10:
Saw your post for the badge, nice neat case! Like the use of partial refresh too! :)

chunkysteveo, 2020-01-10:
...and printing in nylon..... hat off to you!!!