sqfmi/badgy #20: Works with USB power, not with battery?

codepope, 2018-10-01:
Set up badgy on USB power, things behave as expected. Board LED glows flickers blue. Board boots up.

Popped a LIR2450 in with no USB connection and... nothing. Won't wake up, won't reset for new sketches, no LEDs glowing.

Pop in USB with LIR2450 in place, board LED glows bright blue (charging?) and board functions.

Any suggestions as to trouble shooting this?

chunkysteveo, 2018-10-01:
Just received my new (WHITE PCB) version and can echo the same here - Works fine on USB, appears to charge when an LIR2450 is in place (blue LED) and USB plugged in. Can switch on with an LIR2450 and USB connected, and interestingly when I disconnect the USB while still on it still works. Switch off and back on with just the LIR2450 and no power.. Have not looked into testing where power getting to, and also just testing another LIR2450, but this is odd behaviour indeed!  It's like the USB connection trips it on, and without that initial USB trigger, it doesn't power up?

@codepope  - if your board is working with both USB and LIR connected, disconnect the USB and does it still work on the battery? And then when you switch off and on, does it then fail to boot up?

Would be worth people advising which boards they have, I.E. I now have a blue PCB version with programming headers, and now a super cool white PCB with a serial-USB programming chip on the USB (very cool additon!)

codepope, 2018-10-01:
That's *exactly* what I'm seeing. (White PCB too). USB+LIR boots, pull the USB, it carries on running, then switch off and back on and it fails to boot up.

chunkysteveo, 2018-10-01:
hmmmm, maybe the power regulator is not detecting the battery voltage to switch on? Trying another battery here, but sounds like a valid problem.

codepope, 2018-10-01:
Seems to be another symptom - when you do the switch on with USB and then pull the USB the default sketch reports multiple incorrect directions selected when the stick is touched. Plug the USB back in and it reports as normal.

chunkysteveo, 2018-10-02:
Yes(!), did not notice that, but now also seeing that the button message reports randomly when going from USB to BATT, and it tends to report multiple button presses, and incorrect ones. It's actually reporting multiple presses as I type this and I am not touching it. Sounds like the button is not being de-bounced correctly, or something to do with the ground - unless on USB which may be grounding things correctly?? Could also tie in with the battery issue, maybe there is an incorrect ground issue when running on battery, so it won't boot up from battery?

chunkysteveo, 2018-10-04:
@sqfmi - any thoughts on this odd behaviour of the new boards? Noticed the use a new charge IC too, the TP4054.

Flet, 2018-10-05:
Same exact issue here :( 

New white PCB. Tried two different fully charged batteries.

@sqfmi any thoughts?

chunkysteveo, 2018-10-05:
Oh ho.... Huston, we have a problem..! 😣

sqfmi, 2018-10-06:
Sounds like it could be a hardware issue with the new batch, let us investigate the issue and get back to you guys shortly.

chadvavra, 2018-10-06:
I just received 2 white PCB units and can also confirm that they do not boot, or even power the esp on battery power.

sqfmi, 2018-10-07:
Brief update, it seems while on battery power only, you are able to turn it on by resetting it. This can be done by using a metal object to short the reset pin with the metal shield (e.g. paperclip, key, etc.):

[image]

Hacky solution for now, will continue to diagnose the root cause.

codepope, 2018-10-08:
Doesn’t seem to be working for me.

chunkysteveo, 2018-10-08:
I can confirm this hack works for me and boots the ESP8266 on battery power. 

A better connection to "tap" @codepope would be the **lower** pad of the Lipo JST connection (which is ground). Not the upper pad as that is the positive to BATT. The casing didn't seem to give a good reliable ground connection, but turn on the switch, put a wire (I used a dupoint breadboard wire) on the RESET pin imaged above, and then tap the lower JST pad - the ESP should flash it's blue light on boot. (don't tap the left side pads of the JST connection, these don't seem to be tied to ground, just mechanical fixings)

codepope, 2018-10-08:
@chunkysteveo Yes, the JST pad is easier to tap, but still no joy. To ensure I was getting a reset, I applied USB power and ground tapping the reset line did reset.

chunkysteveo, 2018-10-08:
hmmm, not good. Assume battery is fully charged too? Some flake-y issues going on to be sure! Hopefully there will be a fix soon, even if it's a bodge wire fix...!?

codepope, 2018-10-08:
Well, I thought it was, but testing says otherwise. Tried another battery which shows charged and it didn't play ether.

bmv437, 2018-10-09:
As a temporary fix, I added a reset button between the ground pad on the JST connector, and the RST pin on the ESP8266. Also threw on a JST connector and LiPo battery since I didn't have a proper coin-cell size LiPo.

Now I just power it on, and click the reset button to boot the ESP8266.

[image]

chadvavra, 2018-10-10:
I put in a LIR2045 and tried the bodge wire with no luck.  It does work with the USB power though.

sqfmi, 2018-10-10:
@Korenchkin yes! Thank you for the reminder :) The new schematic is in the hardware folder now.

Everyone else, sorry we're still trying to debug the issue :(

Korenchkin, 2018-10-11:
thanks for the schematic!i tried to find the problem,but this makes my head hurt..everything seems okay for starting,i'd try pulling the usb chip and stick with prog header(i'm guessing some leakage to ground),but maybe somebody will make smarter solution..like cap on reset?tried that,but did not work for me..but at least i know what the switch switches...

sqfmi, 2018-10-11:
Removing the CH340C does seem to work! But the reason is unclear at this point... we'll try to do some further investigation with a scope.

chadvavra, 2018-10-11:
I would like to go on the record as saying that the CH340 is the devil's brood.

I flipping hate those things

bmv437, 2018-10-11:
@sqfmi If you're looking for cheap alternatives to the CH340, I'm a big fan of the SiLabs CP2104. Adafruit uses them for all their newer dev boards. High data transfer rates, better driver support than the CH340, and cheaper than an FTDI.
Adafruit has more details on their product page: https://www.adafruit.com/product/3309

Korenchkin, 2018-10-11:
Just a theory,could the ch340 be pulling down gpio0 and/or reset over the 2 transistors?maybe only on start befor it semi-charges its vcc cap?then there are only rx/tx pins,they should not do anything high or low,and that seems all the influence it can do powered off

ibuildrockets, 2018-10-17:
@sqfmi Hmm.. Reset didn't work for me - the LED on the ESP8266 blinks when reset - but nothing. In fact, I get absolutely nothing happenng now :( sketches won't upload and the currently loaded sketch (weather) won't refresh :,( Oh, and the circled component is scortching hot - 
[image]

chunkysteveo, 2018-10-17:
Can you upload a sketch via the USB 2 serial?

g2rHP7prvJqO, 2018-10-17:
> Removing the CH340C does seem to work! But the reason is unclear at this point... we'll try to do some further investigation with a scope.

I am also suffering from this major issue which makes the Badgy basically 99% useless.
I'm strongly considering de-soldering the CH340C but is there any other active/passive component I need to de-solder too to completely fix this issue and avoid any side effects?
Could you please confirm that I can still flash my badgy using a USB-serial adapter connected to pins 1/3 (first two pins top-right opposite of reset pin) once done with the CH340C?

g2rHP7prvJqO, 2018-10-17:
> Yes(!), did not notice that, but now also seeing that the button message reports randomly when going from USB to BATT, and it tends to report multiple button presses, and incorrect ones. It's [...]

I observed that too. It has to do with the fact that the UP/DOWN buttons use the same pins as the Serial TX/RX.

ibuildrockets, 2018-10-17:
> Can you upload a sketch via the USB 2 serial?

No :(

codepope, 2018-10-22:
Any update @sqfmi ?

ibuildrockets, 2018-10-25:
@Korenchkin esptool run - I've finally managed to get it to the point where I can upload the hello sketch - however the display is frozen still on the "weather" sketch :(

ibuildrockets, 2018-10-25:
@Korenchkin I've been using the USB power from the PC, not battery :(

pastorhudson, 2018-10-26:
Just tried my white badgy on battery for the first time, and nothing. It does work when I plug and unplug the usb cable. But power cycling with the switch doesn't work without usb power.

I also experienced phantom Left button and down button presses. They are consistent. I can't use those functions in sketches because they will override other inputs.

sqfmi, 2018-10-28:
Sorry for the delay, here's an update to the issue:

1. Though we're still not exactly sure why the CH340C is causing Badgy to malfunction while on battery power only, removing it does resolve the issue. One option is to remove the CH340C and rely solely on OTA to upload your sketches, like in the original Badgy (all code samples currently have this built-in). If your Badgy requires a hardware flash for recovery, refer to the pin-out diagram below:

[image]

2. As per suggestion by @bmv437 on the CP2104, we tried replacing the CH340C with it and it appears to work fine in our week of testing. We created an adapter board on OSH Park [https://oshpark.com/shared_projects/f7gRYBxi](https://oshpark.com/shared_projects/f7gRYBxi) and it can be ordered for about $0.60. See pics below:

[image]

[image]

We will continue to investigate if there are other viable solutions. In the meantime, we've asked the factory to build some adapter boards for us; when they are ready, anyone experiencing this issue may request a board from us. Going forward, all new Badgys will be using the CP2104 (╯°□°）╯︵ ┻━┻

chadvavra, 2018-10-28:
I found that if you don't want to fully remove the CH340C you can just cut the VCC leg (with a nail clipper in my case) and it will perform as if it's removed.

I figured that if you found another fix it would be easier to bridge that one pin later on.

Huzzah on the adapter boards though.  Those look lovely, and like something I will order a few of for all the other stuff that comes with those dodgy chips.

pastorhudson, 2018-10-29:
Does the CP2104 also address the button issue? Or should I open a separate issue?

sqfmi, 2018-10-30:
@ibuildrockets none of the components should be scorching hot, something may be faulty or damaged for that to occur. Did this always occur, or only after uploading the weather sketch?

@pastorhudson following up on #27 @chunkysteveo has created.

codepope, 2018-10-30:
@pastorhudson taking the ch340c off resolves the button issue (but unfortunately, end of line with my badgy as the epaper apparently cracked in the process.... so can't dig any further).... now to wait for some cp2104 based badgys....

chunkysteveo, 2018-10-31:
@sqfmi - Is there a way to disable or cut the power to the CH340C when on BATT? Would that allow the CH340C to power up on USB and work, but on BATT be powerless and not interfere with the buttons and also maybe boot the ESP8622 chip up? Are the VUSB and VBATT connected, am looking at the schematic and assume no, using the mosfet as a power switch? Or is that the issue, that it is powerless on BATT and causing the issues?

Love the idea of the CP2104 adaptor board and will be keen to get one, but looking to see if we can salvage the CH340C with some bodge wire somehow...?!

chunkysteveo, 2018-10-31:
Or how about bodging on a small slide switch to power off the CH340C? From the comment above - cutting the VCC leg of the chip makes the Badgy work. So could we bodge a "PROG/RUN" switch to essentially power off the CH340C chip to allow it to run? Could we not tie this in with the mosfet that controls when the Badgy is running on USB or BATT power?

Steve

chadvavra, 2018-10-31:
@codepope you can replace the eink really easily.  I changed mine out with a yellow 3 color display.

https://www.waveshare.com/catalogsearch/result/?q=2.9+Inch+E-Paper

JonPrevost, 2018-11-01:
Received 10 white boards all with this battery problem, haven't looked at the button press issue yet. I'm looking into a capacitor (low pass) solution. If effective I'll post my solution.

codepope, 2018-11-02:
@chadvavra Already done that and back up and running today. 

@sqfmi any idea when the adapter board (and next batch of badgys) will be about?

ibuildrockets, 2018-11-07:
@sqfmi - the issue occurred after uploading the weather sketch and resetting the ESP8266 as described in the 'Hacky solution' from October 8th above :(

> Brief update, it seems while on battery power only, you are able to turn it on by resetting it. This can be done by using a metal object to short the reset pin with the metal shield (e.g. paperclip, key, etc.):
> 
> [image]
> 
> Hacky solution for now, will continue to diagnose the root cause.

chunkysteveo, 2018-11-29:
Any updates on the diagnostics, the CP "hack" PCB's, any other fixes, revisions of the PCB @sqfmi ?

sqfmi, 2019-01-02:
Sorry about the delay, we've had trouble finding a fab that could mass produce the CP2014 adapter boards for us. We're in discussion right now with a promising vendor at the moment. We'll keep everyone posted on any new updates and an ETA when it becomes available.

chunkysteveo, 2019-01-07:
That's good, was worried that the project had been abandoned. Do you have the board files available (the add-on board) for anyone that would like to make the CP2104 themselves?

chunkysteveo, 2019-01-08:
Brilliant, i'm sure it will help a lot of guys out. I may even try and solder one up...!

chunkysteveo, 2019-01-08:
Only just spotted your link to the original board project on Oshpark, I could have used that link, apologies!! 

Just out of interest, does anyone know the difference between the CP2104-F03-QFN and the CP2104-QFN? Assume the F03 is a new version, as the "non" F03 chips are cheaper?

ibuildrockets, 2019-01-14:
I've just ordered the PCB from OHS Park - What's the difference between CP2104-F03-**GMR** and CP2104-F03-**GM** - does it matter? Just wanting to get the IC on the way, too :)

sqfmi, 2019-01-15:
They are the same part, the difference is just in packaging (**R** stands for reel)

jcaron23, 2019-02-15:
Hi all,

Are the boards currently sold on Tindie still suffering from this issue, or have they been updated?

Thanks!

sqfmi, 2019-02-28:
@jcaron23 all new boards will come with the CP2104 instead of the CH340C, which does not have the battery issue.

FYI we found a fab to create the adapter boards for us, we are now waiting for the CP2104s to arrive so we can begin testing it out. If they work ok, we will be offering them to those who were affected by the issue.

[image]

chunkysteveo, 2019-02-28:
Awesome news!!

MarquisdeGeek, 2019-03-15:
Is there any alternate fix for those of us with the old board, as shorting the reset pin didn't work for me?

chunkysteveo, 2019-03-15:
What's the difference between the  CP2104-**F03**-QFN and the CP2104-QFN?

@MarquisdeGeek - you can snip the VCC pin of the CH chip to disable it. You will lose the ability to program it via USB though.

MarquisdeGeek, 2019-03-18:
@chunkysteveo - Thanks, but that's a bit drastic for me!

emanuelstanciu, 2019-04-03:
@sqfmi Hi, I have one of the blue PCB models which was bought late last year and I suffer from exactly the same behaviour as described on this post. Any idea what's happening?
Works fine on USB power, but on Battery simply flashes the ESP blue LED and the nothing happens.
Any pointers are greatly appreciated

chunkysteveo, 2019-04-04:
@emanuelstanciu  -  I would recommend starting a fresh issue with this problem. This issue is related to the USB to serial chip not allowing the ESP8266 to boot when on battery. The blue PCB version does not have any USB to serial chip, so whatever issue you are seeing, it isn't to do with this issue. I don't want your issue to be lost in this bigger post!

If the blue light flashes, that would indicate that it has booted and is powered. Something else is not working here?

Steve

pingywon, 2019-07-24:
Any resolution here? I just got my badgy and was under the impression that the white REV B board did not suffer from this problem. I just got some fresh batteries and Badgy doesnt work at all on battery power.

Can someone please advise.

[11 more replies]