sqfmi/badgy #21: Exemple hello compilation

adkp, 2018-10-05:
Hi,
I receive my badgy today :) 
but i can't test  examples, hello or weather 
(test on three computer)
because there is a problem with GxEPD-master library:

Arduino : 1.8.7 (Windows 7), Carte : "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

libraries\GxEPD-master\GxGDEH029A1\GxGDEH029A1.cpp.o: In function `GxGDEH029A1::fillScreen(unsigned short)':

C:\Users\Fabian\Documents\Arduino\libraries\GxEPD-master\src\GxGDEH029A1/GxGDEH029A1.cpp:113: multiple definition of `GxGDEH029A1::fillScreen(unsigned short)'
.......
[compil.txt](https://github.com/sqfmi/badgy/files/2452355/compil.txt)

I need a little help

Thks

pastorhudson, 2018-10-05:
I’m having the same issue.

pastorhudson, 2018-10-06:
I haven't tried battery yet.

adkp, 2018-10-06:
I haven't tried battery yet too

sqfmi, 2018-10-06:
Hello, please use version 2.X of the GxEPD library. Thanks!

closed 2018-10-06 by sqfmi