sqfmi/badgy #26: What have you built?

pastorhudson, 2018-10-26:
This isn't an issue, but not sure where else to talk about this. 

I just finished proof of concept of a Check-in security badge we use at our Church.
https://github.com/pastorhudson/PCOBadge

What have you guys/gals built using Badgy?

lastfuture, 2019-05-20:
I've built a small status monitor for my GitLab issues from work that refreshes hourly. A few coworkers have already told me they'd like something similar.

[image]

lasniko, 2020-01-10:
I built an actual name badge (I am not very creative).
[image]
It is a first generation badgy (blue PCB) with a custom 3D printed frame (Nylon filament) and a magnetic clothing attachment glued to the back. It randomly cycles through various teacher certifications every few minutes using the ESP deep sleep (I tried to do a linear slideshow using system_rtc_mem to count and store the index of which logo to display, but failed. So now it just randomly picks one each wake-up)

boettner, 2020-01-31:
Great Ideas!

I‘m working on a train station dashboard. (I‘d like to know how much time is left for drinking coffee before I‘m ready to go.)

[image]

pastorhudson, 2020-01-31:
@boettner I love that! That's so cool!

bwente, 2020-06-17:
I just made a desktop display last night.

[image]

I wish on/off switch was a bit more accessible for the case. Can the button be used to do a reset?

https://www.reddit.com/r/esp8266/comments/hawrqr/dad_jokes_for_fathers_day/

chunkysteveo, 2020-06-18:
@bwente  - looks gorgeous! (funny joke too!!) You could re-purpose the main button to be used in the sketch to trigger an ESP_reset. You'd need to check for it being pressed in the loop, and then trigger a function to call the ESP to reset.

chunkysteveo, 2020-06-18:
@bwente  - this example may help you with your wrapping issue:

`char str[] ="Hello World! Here is some text. I need them to split into multiple lines without breaking the words.";
const int wantedLength = 20;

const char * split (const char * s, const int length)
  {
  // if it will fit return whole thing
  if (strlen (s) <= length)
    return s + strlen (s);

  // searching backwards, find the last space
  for (const char * space = &s [length]; space != s; space--)
    if (*space == ' ')
      return space;
    
  // not found? return a whole line
  return &s [length];        
  } // end of split
  
  
void setup ()
  {
  Serial.begin (115200);
  Serial.println ();

  const char * p = str;
  
  // keep going until we run out of text
  while (*p)
    {
    // find the position of the space
    const char * endOfLine = split (p, wantedLength);  
    
    // display up to that
    while (p != endOfLine)
      Serial.println (*p++);
      
    // finish that line
    Serial.println ();
    
    // if we hit a space, move on past it
    if (*p == ' ')
      p++;
    }
  }  // end of setup
  
void loop () { }`