sqfmi/badgy #27: Button Phantom Pressses

pastorhudson, 2018-10-30:
I have a [working sketch](https://github.com/pastorhudson/PCOBadge) that connects to an api and pulls check-in information and displays it on the badge when I press the right toggle button. I'd like to also display the check-in barcode when I press left.

When I started testing this I found that the left button and down button are triggered all the time.
After I click right it immediately triggers left and down. I've increased the milliseconds on the debounce but it had no effect. It didn't change how fast it was triggered either.

Either my button is defective or something else is interfering?

Is anybody else having this issue?

Has anybody solved it?

chunkysteveo, 2018-10-30:
I assume this is related to [issue20](https://github.com/sqfmi/badgy/issues/20) when running on battery (if it is reset and boots) the buttons get triggered randomly. If you run on USB, it should work fine. Running on battery (if you have it working on USB, then unplug) the button presses will appear..... correct?

pastorhudson, 2018-10-30:
I get the button presses when plugged into USB and when unplugged. I set a Serial.println to trigger when a button is presses and it triggers on USB. It also triggers the screen change I setup when It's off USB.

chunkysteveo, 2018-10-30:
You have the white version PCB board? Looks like the USB2Serial chip is interfering.

sqfmi, 2018-10-30:
The serial TX/RX pins GPIO1 & GPIO3 are shared with the direction buttons left & down (refer to the [schematic](https://github.com/sqfmi/badgy/blob/master/hardware/schematic.pdf)). This is why you will receive "phantom" presses when using the serial terminal. The USB-Serial functionality is only meant for uploading sketches.

If your application requires a serial terminal, you can remove the down & left button (GPIO 1 & GPIO 3) from the polling loop. For data input/output applications, we recommend using WiFi (e.g. http/websockets/etc.)

chunkysteveo, 2018-10-30:
Ahhh yes I recall now - the TX/RX pins. @pastorhudson  check your sketch for any "Print Serials" that may be going up/down the pins and falsely triggering a button press.... totally forgot about that @sqfmi , not the USB2Serial issue causing this!

pastorhudson, 2018-10-30:
Ok maybe I left some Serial code in the sketch and that caused it to trigger when it was running on battery. I'll double check my code, test and report back.

pastorhudson, 2018-10-30:
That was the problem! I had missed some Serial commands. Thank you!

pastorhudson, 2018-10-30:
I spoke too soon. I'm using the method of starting it on usb power and then unplugging it to run on battery. (I have the white version)
It was working fine plugged into a USB power source(Wall Wart not computer) and then when I unplugged the usb power it started doing the left down presses again.

chunkysteveo, 2018-10-31:
@pastorhudson - on Battery the CH340C seems to be interfering with the ESP8266 and the button presses. To make the 5-way work they have had to share GPIO ports, and the TX/RX of the CH340C are also the UP and DOWN button press when pulled low. I would advise getting rid of your "case 0,1 and 4" in the button "switch case" in the loop, and remove all serial code lines (just looked at your githhub code).

pastorhudson, 2018-10-31:
@chunkysteveo Thanks for the ideas. This is the code I tested for the comment where I reopened the issue. I didn't push to github because it has some other dev stuff I'm not ready to push yet.

```
/*  PCOBadge is a fun project by Ron Hudson.
    PCOBadge pulls an individual's checkin info from the Planning Center Check-ins api and displays it on an e-paper display.
    It lives at https://github.com/pastorhudson/pcobadge
    It is Liscenced under the:

  MIT License

  Copyright (c) 2018 Ron Hudson

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
  SOFTWARE.

*/

/* Libraries You Need */

/* e-paper display lib Version 3.x
   https://github.com/ZinggJM/GxEPD/releases */
#include <GxEPD.h>
#include <GxGDEH029A1/GxGDEH029A1.h>
#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>

/* include any other fonts you want to use https://github.com/adafruit/Adafruit-GFX-Library */
#include <Fonts/FreeSansBold9pt7b.h>
#include <Fonts/FreeSansBold12pt7b.h>
#include <Fonts/FreeSansBold18pt7b.h>
#include <Fonts/FreeSansBold24pt7b.h>

/* lable.h is a byte encoded version of label.bmp Label Image */
#include "label_cbc.h"
#include "apiStatus.h"
#include "labelMenuBeta.h"

/* WiFi  libs*/
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <WiFiManager.h>

/* Misc Libs */
#include <ArduinoJson.h>
#include <Time.h>
#include <TimeLib.h>
#include <base64.h>

/* Always include the update server, or else you won't be able to do OTA updates! */
/**/const int port = 8888;
/**/ESP8266WebServer httpServer(port);
[... 247 lines cut ...]
  if (strptime(buff, "%y-%m-%dT%H:%M:%SZ", &tm) == NULL);
  else {
    time_t t = mktime(&tm);
    String(tm.tm_mday) + " " + String(tm.tm_mon);
    String time_stamp = String(dayShortStr(tm.tm_wday)) + " " + String(monthShortStr(tm.tm_mon + 1)) + " " + String(tm.tm_mday) + " " + String(tm.tm_year);
    return time_stamp;
  }
}

void showText(char *text)
{
  display.setRotation(3); //even = portrait, odd = landscape
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(h4);
  display.setCursor(10,70);
  display.println(text);
  display.update();
}
```

Sorry for the big dump. I removed all Serial lines. And on battery Left and Down cases are still triggered. It seems from a comment in #20 that this is indeed related to the CH340C? This isn't a mission critical thing I'm building. So for now I can use Up and Right. But I'd love to get the adapter. 

PS
@sqfmi We all know this stuff is really hard, and I am super thankful that you built Badgy. It's such a fun platform at a really good price. I can't wait to buy the next better version 👍

chunkysteveo, 2018-10-31:
That's a lot of code! But all looks fine. My thought i didn't say, Which is what you have already said - was to use the other buttons as a work around. It's a cool script by the way, watched your video!

pastorhudson, 2019-05-29:
Closing this up unless anybody else is experiencing it

closed 2019-05-29 by pastorhudson