sqfmi/badgy #28: @sqfmi Hmm.. Reset didn't work for me - the LED on the ESP8266 blinks when reset - but nothing. In fact, I get absolutely nothing happenng now :( sketches won't upload and the currently loaded sketch (weather) won't refresh :,( Oh, and the circled component is scortching hot -

JonPrevost, 2018-11-05:
[image]

_Originally posted by @ibuildrockets in https://github.com/sqfmi/badgy/issues/20#issuecomment-430591985_

JonPrevost, 2018-11-05:
Left on with websocketDraw example and I'm having the same problem. This is very hot and nothing is working now. It was left on for 2 days. Will not reset. The CH340C is still in-place, unmodified board.

ibuildrockets, 2018-11-17:
I've tested the hot component. Ambient is 25ºC (77ºF) and the component is at 89ºC (192.5ºF)

Again, no life in the badge :(

[image]

sqfmi, 2019-01-02:
The inductor shouldn't be that hot ♨️ ! Please try replacing it with an equivalent 68uH 1210 inductor, or reach out to us with your order number and we'll send you a replacement board.

ibuildrockets, 2019-01-06:
Oh that's super! Tindie order number is 122802 under the name Sascha Grant.

ibuildrockets, 2019-01-15:
@sqfmi Thank you for shipping a replacement PCB :) I've not transferred the display across yet. I've just ordered the PCB from OHS Park - What's the difference between CP2104-F03-GMR and CP2104-F03-GM - does it matter? Just wanting to get the IC on the way, too :)  Figure that I'll get that on the board first.

sqfmi, 2019-05-18:
@ibuildrockets they should both be the same, just different packaging. Closing this issue now.

closed 2019-05-18 by sqfmi