sqfmi/badgy #29: Weather sketch is broken

davidharpe, 2019-01-13:
Hi, the weather sketch is broken. Forecast request dies with a json parsing error. Not sure where the problem is...tried and failed to figure it out. Best I could tell it looks like stuff is left in the buffer after the first current conditions call. Also setting imperial units does not change the conversions or the displayed units deeper in the code, which makes setting the constant somewhat pointless:   
    259 display.print(String((int)(wind*3.6))+"km/h");   
    268 display.println(String((int)current_temp) + "C");

JoergSn, 2019-01-26:
Regarding the forecast weather, sometimes the json is too large for the Badgy to handle it. Found a solution:

// use **cnt=25**; might be enough for three days etc.
// it's undocumented, but it works
String url = "/data/2.5/"+type+"?id="+CITY_ID+"&units="+UNITS+"**&cnt=25**&appid="+API_KEY;
// String url = "/data/2.5/"+type+"?id="+CITY_ID+"&units="+UNITS+"&appid="+API_KEY; // we get often to much data for this little thing, cannot handle it

mrWheel, 2019-05-10:
Latest version of arduinoJSON (6.n) does not work with this sketch.
Should be updated!

sqfmi, 2019-05-18:
Weather example has been updated. Closing this issue for now.

closed 2019-05-18 by sqfmi