sqfmi/badgy #30: Blue PCB not starting on battery, USB is fine

emanuelstanciu, 2019-04-04:
Discussed this on #20 , pasting here the conversation relevant for this new issue:

@sqfmi Hi, I have one of the blue PCB models which was bought in July last year and I suffer from exactly the same behaviour as described on this post. Any idea what's happening?
Works fine on USB power, but on Battery simply flashes the ESP blue LED and the nothing happens.
Any pointers are greatly appreciated - if it helps: Tindie Order #115523

Thanks @chunkysteveo for the quick response

emanuelstanciu, 2019-04-04:
Please let me know if you want me to provide any details to get this progressed, not sure how to move forward with this one

chunkysteveo, 2019-04-04:
Are you able to confirm the voltage of the battery? With a multi-meter?

emanuelstanciu, 2019-04-04:
I am, will do this as soon as I get home tonight. Cheers

emanuelstanciu, 2019-04-05:
3.27V - brand new battery
I can offer something that might bring some light to this story; I am using a CR battery (non-rechargeable), not a LIR that I've seen around various articles. Can this be an issue, in terms of Amperage? 
I tried 3 batteries already, same effect

chunkysteveo, 2019-04-05:
You HAVE to use a LIR if you plug it into USB for power, as the board has a charging circuit and WILL try to charge it. The voltage is fine though, not the issue.

So when you have it plugged into USB, how do you know it works? When on battery and you flick the switch to on - how do you know it's not working?

emanuelstanciu, 2019-04-05:
I ordered a LIR and will test it over the weekend. But it should still work on battery, does a CR provide the right amount of current?
on USB, ESP flashes blue and then a few seconds later the sketch is loaded (hello.h) and I see the screen updating. I can also put it in OTA mode and can access the URL from my laptop
on battery, there's the same brief flash of the bllue LED on ESP and then absolutely nothing happens/works
I can do a quick video if you want to see it, let me know and thanks for your help with it

chunkysteveo, 2019-04-05:
The current will be fine between a normal CR3032 and an LIR3032, no issue. 

That is odd behaviour. When you switch it on when on battery - it would be worth you checking your WiFi networks on your phone/PC - can you see a Badgy access point? Or an ESP access point? I recall something similar to me where I wasn't totally sure it was working or not, but it turned out I needed to connect to it via phone, enter my WiFi credentials of my home WiFi, and then it worked fine.... Worth a go?

Steve

sqfmi, 2019-04-06:
Hi @emanuelstanciu,

You **MUST** use the LIR2450 battery as stated on the Tindie product page.

CR batteries (non-rechargeable button cell type) have a nominal voltage of 3V, and will not sufficiently power badgy, as the ESP8266 requires 3.3V. In addition, the LDO regulator expects a higher voltage so it can step down to 3.3V.

LIR batteries (rechargeable button cell type) have a nominal voltage of 3.6V; this gets stepped down by the LDO to 3.3V.

The blue LED flash you see is the ESP8266 trying to boot up on insufficient voltage input.

Thanks @chunkysteveo for chiming in to support :)

chunkysteveo, 2019-04-06:
Ahhh...i didn't know there was such a difference in the CR and LiR?! And apologises- i said the 2032, but it's the bigger 2450 versions.

emanuelstanciu, 2019-04-06:
Thank you both. The LIR2450 arrived today and indeed it worked without an issue. Tanks again for a great product.

closed 2019-04-06 by emanuelstanciu