sqfmi/badgy #31: Screen Alignment Issue - Rev 2B

bmlzootown, 2019-05-09:
Got the new Rev 2B board the other day, but I was sick so I didn't get to tinker right away. Pulled it out earlier today and, upon uploading the hello example (slightly modified with my initials instead), I realized that the screen was misaligned. Even the ["Badgy AP" screen](https://i.imgur.com/f3yGCKx.jpg) seems to be misaligned. 

I'm new to... well, all of this, so I apologize if my description isn't quite apt. Is there a simple solution that I'm just completely unaware of?

Thanks!

sqfmi, 2019-05-10:
Could you check to make sure the screen's ribbon cable is properly attached to the connector? This happens sometimes when it's loose or misaligned.

bmlzootown, 2019-05-10:
No luck -- the [end result](https://i.imgur.com/uvBQaki.jpg) still appears to be the same, even when [completely removed and then reconnected](https://i.imgur.com/Gq2NkEI.jpg).

sqfmi, 2019-05-10:
The connection looks good. The display appears to be not refreshing the whole screen, but just the misaligned section. The only time we've encountered this is when the ribbon cable is not aligned or loose... can you provide more details on your setup so we could try to help troubleshoot?

mrWheel, 2019-05-10:
Help!
I'm having the same issue!
How do I remove/re-insert the ribbon cable without breaking my Badgy?
[image]
[image]
[image]

sqfmi, 2019-05-10:
Could you also try power cycling a few times, and using a different USB cable/port?

mrWheel, 2019-05-10:
This is what I did:
- replaced the battery
- changed USB cable
- switched on/off with USB cable attached
- switched on/off without USB cable with old and with new battery

No change :-(

sqfmi, 2019-05-10:
@mrWheel what type of battery is being used?

mrWheel, 2019-05-10:
LIR2450’s

sqfmi, 2019-05-11:
One more thing to try, could you remove all power sources (i.e. battery and USB) and wait a few seconds, then with the switch already in the **ON** position, plug in the USB cable?

This is weird behavior and new to us, we have not been able to deterministically reproduce the same misaligned results. We will continue to investigate this issue.

bmlzootown, 2019-05-11:
Did the above: removed all power sources, flipped the switch to the ON position, and then plugged in the usb cable. The entire screen updated this time. 
[image]

Turning the switch back OFF before re-plugging the cable reproduces the partial screen update issue. 
[image]

Interestingly enough, after doing the above (returning to a partial screen update), it then took me a good 3-4 more tries to get the full screen to update again using the process described in your last reply/

mrWheel, 2019-05-11:
Ok, did that.
-Battery removed
- USB disconnected
- switch on .. off .. on .. off .. on
> nothing happens on the screen though
Result is the same mis-alignment

peteargles, 2019-05-18:
Figured I should log in and chime in. 
Mine arrived last week and I had the same screen alignment issue. 
I removed and reinserted the ribbon cable and it's been perfect ever since.

Lupuseli, 2019-05-28:
I have experienced the same issue. However, it manifests depending on the power supply: whenever I have used a power supply/charger which had Qualcomm’s Quick Charge or Samsung’s Adaptive Fast Charging (both exceed the USB spec’s voltage parameters) I noticed the screen alignment issue. The solution was using a different power supply (standard USB-PD, like the ones used to power Raspberry Pi's). Goes back to normal after one power cycle.

Jonzay, 2019-05-31:
I also experienced the same issue while I was uploading code via USB from my PC, with the LIR2450 battery connected. It went away when I removed the battery. I've only just reinserted the battery (tested fine running the code I had uploaded), so I will see if I can get the issue to reoccur tonight and create specific steps for replicating the issue.

clsn, 2019-06-02:
I'm having the same problem, and I don't even have any battery installed yet.  I'll try disconnecting from power with the switch ON, in the hopes that it'll reset everything as it seems to have for those above.

afonso360, 2019-06-15:
Had the same issue, it comes and goes, when i unboxed it, it didn't update the full screen, a few hours later it updated the entire screen. Today, partial update again, removed and reinserted the ribbon cable, and its ok now.

noxo, 2019-06-21:
I have the same issue, have been powering the device via USB only (laptop USB-port). Not sure, but it seems that removing the power source (USB-cable) for 30 secs or so, and then reconnecting the power source seemed to fix the thing.

Jonzay, 2019-06-28:
> I will see if I can get the issue to reoccur tonight and create specific steps for replicating the issue.

Well, it's finally tonight :P

I got the Badgy out to upload new code (to try to get it onto WPA2-Enterprise networks), had it switched off but with the LIR2450 battery in it, plugged in the cord, switched the switch on and experienced the issue. I could only get full screen updates back again by removing the USB, switching the switch off, removing the battery, switching the switch on and then plugging in the USB cable. It is behaving itself again.... for now.