sqfmi/badgy #33: Badgy does not reset after flashing via USB

lastfuture, 2019-05-18:
Hello,

my Badgy Rev 2B does not reset itself after successfully flashing it via the Arduino IDE through USB. The upload log reads "Hard resetting via RTS pin..." but the display will remain in the state it was in during upload unless I either toggle the on/off switch or I powercycle the badge by unplugging and plugging the USB cable.

The behavior is the same with all example sketches as well as my own test sketches down to the very basics of simply displaying a text without any Wifi capability.

"Reset Method" is set to "ck", but neither of the other available options changes the behavior.

What do I have to do to make Badgy reset itself upon successful upload via the Arduino IDE?

sqfmi, 2019-05-18:
Hi @lastfuture ,

Badgy uses the [two transistor circuit](https://tttapa.github.io/Images/thumbs/auto-reset_auto-program.png) for autoreset using the DTR/RTS lines. You can select '**nodemcu**' for 'Reset Method' and it should perform the reset after upload.

If that is not working either, you can check the [green PCB](https://raw.githubusercontent.com/sqfmi/badgy/master/website/img/badgy_fix_2.JPG) in the back to see if there are any issues with the DTR/RTS connection spots.

The power switch also serves as reset, but ideally it should work through the autoreset mechanism.

Xaldimo, 2019-05-18:
I have noticed with version 2.5.0+ version of the ESP8266 board driver the reset doesn't work any more. Either using the ESP8266 Boards or NodeMCU 1.0 (ESP-12E Moldule)

lastfuture, 2019-05-19:
I couldn't make a difference using the software options alone and after setting the board to NodeMCU 1.0 the reset option disappeared altogether. Neither with the "Reset Method" set to nodemcu nor with the board set to NodeMCU 1.0 does Badgy reset after an upload.

The soldering on the connection spots looks fine. I have no oscilloscope but I was able to check for continuity and the tops of the DTR/RTS pads connect to the top and bottom 10k resistors respectively that are directly right of the green PCB (as parts of the two transistor circuit you mentioned I assume).

lastfuture, 2019-05-19:
I can confirm that after downgrading from version 2.5.1 to version 2.4.2 I'm now seeing autoreset working as expected. I had to change my sketch though because I was using SSL certificate fingerprints which 2.4.2's WiFiClientSecure doesn't seem to support yet.

sqfmi, 2019-05-22:
Closing as this issue relates to the ESP8266 Arduino library, adding a note to the troubleshooting guide.

closed 2019-05-22 by sqfmi