sqfmi/badgy #34: List of optimal board settings?

lastfuture, 2019-05-18:
Hi,

after setup of the board in the Arduino IDE the board settings got populated with what looks like defaults. Some of the defaults seem wrong, like the Flash size of 512k (no SPIFFS). During upload the correct flash size of 4MB seems to be negotiated, and the upload works fine, but this made me start wondering what else might not be on the correct setting. I'd like to set up Badgy as well as I can. By playing around with them though I nearly bricked my badge and was very lucky that it recovered to a state in which it would accept new firmware with the reverted settings.

Here's a screenshot of the default settings. Can anyone provide the "correct" settings tailored for Badgy or are these already the best they can be?

[image]

Xaldimo, 2019-05-18:
You want to select "NodeMCU 1.0 (ESP-12E Module)" as mentioned in https://github.com/sqfmi/badgy/tree/master/examples Setup section.

[image]

lastfuture, 2019-05-18:
Oh thank you, this looks much more reasonable.

closed 2019-05-18 by lastfuture