sqfmi/badgy #35: Free GPIO pins?

lastfuture, 2019-05-19:
I've got ideas for Badgy that require me to connect to a GPIO pin (e.g. a 1-wire digital temperature sensor). I see that GPIO9 seems to be free. Can I solder to it or is it used internally by the package? Are there other free pins that I'm missing? If there aren't any, I can sacrifice directions of the joystick, right?

sqfmi, 2019-05-22:
GPIO9 is used by the ESP-12 for internal flash memory, so using it will cause resets. You can sacrifice the direction pins to use as GPIO, but note that GPIO1,3 are shared with TX/RX, so serial cannot be used concurrently with these pins. GPIO10 is also input only, so it cannot act as output.

gerwitz, 2020-07-21:
Curious if you went anywhere with this, I'd also like to add a temp sensor.

sqfmi, 2020-07-21:
You can sacrifice the direction pins to use as GPIO, but note that GPIO1,3 are shared with TX/RX, so serial cannot be used concurrently with these pins. GPIO10 is also input only, so it cannot act as output:

[image]