sqfmi/badgy #36: Failed to connect to ESP8266: Invalid head of packet (0x62)

AlanCunningham, 2019-05-29:
I was having a go at putting micropython on the board, however mistakenly attempted to upload the "esp8266-20190125-v1.10.bin" micropython file via the OTA method rather than USB.

I'm now trying to install the default badgy.bin file via Arduino to start again however get the following exception:

```
Sketch uses 333532 bytes (31%) of program storage space. Maximum is 1044464 bytes.
Global variables use 37136 bytes (45%) of dynamic memory, leaving 44784 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\...\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/.../Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/.../Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0x6C)
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0x6C)
```

Has anyone come across this before?  Can I reset Badgy or still put micropython on it?

sqfmi, 2019-05-29:
Try using the esptool and erasing the flash. You may also have to toggle the on/off switch at the right time if you are unable to get it into upload mode.

cunnia06, 2019-05-29:
Thanks for the reply.  Regarding toggling the on/off switch at the right time to force into upload mode - is there a specific time to do this?  E.g. a few seconds after plugging the USB in?

AlanCunningham, 2019-05-29:
Thanks for the reply.  Regarding toggling the on/off switch at the right time to force into upload mode - is there a specific time to do this?  E.g. a few seconds after plugging in the USB?

chunkysteveo, 2019-05-29:
It all depends on the sketch. The on/off switch won't put it into upload mode - but most of the badgy sketched use the deepsleep, so once it's booted and done "whatever task" it will shut down. I found a little more reliable is to hold the joystick button down on boot. This (usually - again, depends on the sketch) will put it into OTA Upload mode and not set it to go to sleep. It will be sat waiting for a HTTP upload, but more importantly will be up and running and able to accept Tx Rx signals from the USB chip.

AlanCunningham, 2019-05-29:
Success!  Using esptool to erase the flash did the job.  In case anyone finds themselves in the same situation as me, here's the steps I took to erase the flash and put micropython on the board:

- Plug the board in with the On/Off switched to Off
- Run `$ esptool.py --port /dev/tty.SLAB_USBtoUART erase_flash` (Your port may vary)
- You'll see a `Connecting........_____` message
- During this message, hold down the joystick button and turn the switch to On.  Then release the joystick button.  Doing this near the beginning of the "Connecting" message worked for me.

Full output:
```
$ esptool.py --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py v2.6
Serial port /dev/tty.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:84:63:66
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.1s
Hard resetting via RTS pin...
```

Installing micropython was then a similar process:
```
$ esptool.py --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash --flash_size=detect 0 esp8266-20190529-v1.11.bin
esptool.py v2.6
Serial port /dev/tty.SLAB_USBtoUART
Connecting........_____....._
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:84:63:66
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 617880 bytes to 402086...
Wrote 617880 bytes (402086 compressed) at 0x00000000 in 9.6 seconds (effective 517.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Badgy now broadcasts the Micropython AP successfully.  Thanks very much for your help!

chunkysteveo, 2019-05-29:
Awesome, glad the little joystick hack worked for you!

closed 2019-05-29 by AlanCunningham