sqfmi/badgy #37: Can Badgy measure the voltage of the battery?

mrWheel, 2019-06-05:
Is it possible (with minor modifications, if needed) to measure the voltage of the battery?

chunkysteveo, 2019-06-05:
You can - jumper the ADC pads on the back and that will give you the voltage as an analogue readout using the built in ADC of the ESP8266 chip.

sqfmi, 2019-11-18:
Moved to Wiki https://github.com/sqfmi/badgy/wiki/Can-Badgy-measure-the-voltage-of-the-battery%3F

closed 2019-11-18 by sqfmi

chunkysteveo, 2022-09-26:
Adding to this all the way in 2022(!) Bridging the ADC solder jumper will only measure up to 3.3V as the voltage divider is 220k/100k - to get a more accurate battery voltage you should bridge the ADC jumper with a 100k resistor, which will bring the recorded ADC value closer to 4.2V when at or near 1V seen on the ADC ( voltage divider will be  = 320k / 100k).