sqfmi/badgy #38: What is the meaning of the blue LED (not the one on the ESP8266)

mrWheel, 2019-06-09:
Hi,

When I apply USB power to the Badgy a blue LED lights up (not the LED on the ESP8266). I have seen before that after a while the blue LED blinks but recently is only fades and it seems the battery is not charged ...

What is the LED  supposed to do?
What is the information it gives when on, off, blinking, and faded?

chunkysteveo, 2019-06-10:
The LED is part of the TP4054 charge circuit, and is attached to the CHRG (or STAT) pin of the TP4054 chip from Vusb and a 2K resistor. That pin has a tri-state, low, high, and high impedance. It's a tricky pin and tricky state to understand, even I don't fully "get it". But the best write up I can find from the spec sheets is this:

CHRG is the open-drain status indication output. When the charger is in the charging state, CHRG is pulled down to low level. After the charging is completed, the CHRG pin is in the high-impedance state; if the battery is not connected, the BAT pin voltage is inserted when the charger is inserted. At about 4.6V, the indicator will flash indicating a missing battery or bad battery contact.

When the charger charges the battery, the CHRG pin is pulled low by the internal switch to indicate that the charging is in progress; when charging is completed, the CHRG pin is in a high-impedance state and the indicator light flashes when the battery is inserted and the battery is disconnected, missed battery or bad battery contact.

So to answer your setup - a flashing LED would indicate that there is no battery present when plugged into USB power. If your battery is not charging, there may be an issue with the TP4054 chip, or the battery.  I have a similar issue with an old version of the badgy - replaced the TP4054 and still won't charge. I need to start looking at the circuit paths in more details to see where it may have failed.

Hope that helps.

sqfmi, 2019-11-18:
Moved to Wiki https://github.com/sqfmi/badgy/wiki/What-is-the-meaning-of-the-blue-LED-(not-the-one-on-the-ESP8266)

closed 2019-11-18 by sqfmi