sqfmi/badgy #40: Unable to wake from DeepSleep

liorbp, 2019-06-18:
Hi there,

I just realized I did something stupid by calling some function directly from setup() that ends with the sequence below and now I'm not able to flash a new sketch :(
```
delay(1000);
ESP.deepSleep(1 - 1000000UL);
```

I noticed this in the README:
> Note that if the ESP chip is in a deep sleep then it will not automatically respond to the upload request and you may see something like "error: espcomm open failed" when trying to upload. A manual restart of the ESP chip, using the power slide button, just prior to upload gives you a small window of time to upload.

I was trying everything, switching the button, run the flash command using esptool in a loop and connecting the badgy while it runs. Nothing I did allowed me to re-flash it.
My attempts will always end with:
```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

Any other way to wake up the badgy to allow flashing?

chunkysteveo, 2019-06-18:
Have you kept the OTA Sketch upload part from the example sketches? This is where you hold down the centre button of the joystick to get it into upload mode from a web interface. If you do that, it will keep the ESP8266 alive while it waits you to upload a BIN file. Ignore that, but now the ESP8266 is no longer running the sketch so will be able to be flashed. Can you post your whole code to this to see at what point you have added the sleep part?

liorbp, 2019-06-18:
Unfortunately, I didn't keep the OTA Sketch upload section (as I said, stupid).
The sketch was something like the below code... is it game over to my badgy? :(

```
GxIO_Class io(SPI, SS, 0, 2);
GxEPD_Class display(io);

void setup()
{
  Serial.begin(115200);
  display.init();
  myfunc();
}

void loop()
{
}

void myfunc()
{
  delay(1000);
  ESP.deepSleep(1 - 1000000UL);
}
```

chunkysteveo, 2019-06-19:
No, it's not game over... you have a delay in there of a second, and the boot up from sleep also takes some time. The quick way will just be a case of trying to upload your sketch to the board, and timing the off/on right so it (the USB 2 Serial chip) catches the ESP8266 in it's boot up and sketch loading state. What revision board do you have?

A longer and harder way would be to manually upload some code to the ESP8266 using some Dupoint cables and connecting up power, ground, RT, TX and setting the GPIO appropriatley. But you would need an external serial programmer, even a spare Wemos D1 mini or Node MCU would do. As you have spotted in your link, the pins would need to be:

UART Download Mode (Programming): GPIO 0 -  LOW, GPIO 2 - HIGH, GPIO 15 LOW.

This will force the chip into programming mode when you power it up and you are not relying on the Badgy USB2Serial chip catching it in time. Then you can upload at your leisure.

Steve

closed 2019-06-24 by liorbp

chunkysteveo, 2019-06-24:
sorted??