sqfmi/badgy #41: 'WiFiManager' was not declared in this scope

SciaticNerd, 2019-06-23:
Hi there! I'm trying this out for the first time... all of it.  I'm following the 'getting started' type info
1) First from the 'compiling examples' info at: https://github.com/sqfmi/badgy
2) and then following the setup section in: https://github.com/sqfmi/badgy/tree/master/examples
3) before getting the following error when trying to compile:
https://github.com/sqfmi/badgy/blob/master/examples/hello/hello.ino 

Just copied the text from the hello section and pasted it into the IDE and changed it to be my handle. **Is there another way this is supposed to be done?**

----------------------- begin error text copied from Arduino IDE ------------------
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

hello:111:25: error: variable or field 'configModeCallback' declared void

 void configModeCallback (WiFiManager *myWiFiManager){

                         ^

hello:111:25: error: 'WiFiManager' was not declared in this scope

hello:111:38: error: 'myWiFiManager' was not declared in this scope

 void configModeCallback (WiFiManager *myWiFiManager){

                                      ^

C:\Users\Admin\Documents\Arduino\Badgy\Hello\hello.h\hello.ino: In function 'void setup()':

hello:51:31: error: 'configModeCallback' was not declared in this scope

     wifiManager.setAPCallback(configModeCallback);

                               ^

exit status 1
variable or field 'configModeCallback' declared void

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
-------------------- end error dump -------------------

Using Arduino 1.8.9 (Windows Store 1.8.21.0)

Thanks for any help in advance!

Respectfully,

SciaticNerd

SciaticNerd, 2019-06-23:
Went back and re-ran this with verbose on for Compiling and 'more' set to on... Output follows, in case it helps...

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Admin\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Admin\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_820327 -warnings=more -build-cache C:\Users\Admin\AppData\Local\Temp\arduino_cache_112899 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -verbose C:\Users\Admin\Documents\Arduino\Badgy\Hello\hello.h\hello.h.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Admin\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Admin\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_820327 -warnings=more -build-cache C:\Users\Admin\AppData\Local\Temp\arduino_cache_112899 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\Admin\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -verbose C:\Users\Admin\Documents\Arduino\Badgy\Hello\hello.h\hello.h.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Admin\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2
Using core 'esp8266' from platform in folder: C:\Users\Admin\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2
Detecting libraries used...
"C:\\Users\\Admin\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Admin\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\Admin\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\Admin\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_820327/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Admin\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp [... cut ...]
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Admin\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer 
Using library ESP8266HTTPUpdateServer at version 1.0 in folder: C:\Users\Admin\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPUpdateServer 
Using library WiFiManager at version 0.14 in folder: C:\Users\Admin\Documents\Arduino\libraries\WiFiManager 
exit status 1
'configModeCallback' was not declared in this scope

chunkysteveo, 2019-06-23:
Please don't paste all the debug code, that's a huge amount of waffle! But the first lines should help you - it's showing 4mb, no spiffs. You need to change the board options to 1mb spiffs to allow you to save config info such as wifi data etc.

I am assuming you have WiFi Manager library installed for your Arduino IDE setup, but worth asking?

Steve

SciaticNerd, 2019-06-23:
So it DOES matter... compiled just fine using git.... glad to have been able to learn and will share this for other folks just starting out:

install git
Create a folder 
launch a command line
navigate to the folder you just made
git clone https://github.com/sqfmi/badgy
this will create a local copy of folders for badgy to the local computer
now open the hello.ino file in the arduino IDE
change the name from "badgy" to your chosen name.

Those with newer models of the Badgy IoT can use the Arduino IDE to compile and it will push the update to the connected device. 

If uploading the code using WiFI
From inside the Ardunio IDE...
click on Sketch > Export compiled binary (this is the way I know this works as far as I know right now)
go find the outputted file in 
C:\Users\<YourAccountNameHere>\AppData\Local\Temp\arduino_build_<some number> <--- Look for the folder with the closest time to when you clicked Compile
locate the file called  'build.ino.bin'
Now you're ready to install this to your badgy iot device!

Connect the Badgy to power (battery or a USB cable connected to a computer or power source)
Holding the badgy so the screen is facing you, move the switch into the up position (OFF)
Press and hold the joystick button like a button and move the switch down (ON)
let go of the button when the display changes to words about how to connect

If this is the first time the badgy is being set up, its WiFi will need to be configured
follow the on screen prompts on the device

If WiFi has already been configured, the badgy will display a URL
Now open a web browser and navigate to the address displayed on the Badgy's screen.
Click on "Choose File" on the web page portion of the screen
Select the .bin file located earlier in the export binary step and click "Open"
Click update and watch it go! (the page will be 'not found' after the device reboots)
The Badgy will now update and reboot itself!

Like what you see on the screen? Great! 
The Badgy will run on either a LIR2450 coin cell or 1S LiPoa battery - just make sure the battery is inserted matching the symbol on the little metal piece that holds it 
(in other words - if the metal shows a plus, then make sure the matching shape is facing you when inserting the battery)
So flip the power switch to the down position (OFF) to save power and leave the eInk display showing the "Hello my name is" message!

My name needs to be centered in order to fit, but it works now, so I'll close the ticket.
tl;dr: Clone the example files with git instead of copy and pasting and things compiled just fine.

Thanks much!
Steven
@SciaticNerd

SciaticNerd, 2019-06-23:
> Please don't paste all the debug code, that's a huge amount of waffle! But the first lines should help you - it's showing 4mb, no spiffs. You need to change the board options to 1mb spiffs to allow [...]

Hadn't thought it would be so ginormous... Deleted. 
Definitely had the libs installed... being so new, I tried to copy/paste from github instead of using 'git clone'
Used git > changed name > compiled without incident... 

I still have no idea what I'm doing, but still succeeded, so I'm counting it a win! :-D

closed 2019-06-23 by SciaticNerd

chunkysteveo, 2019-06-24:
Glad you sorted it. A GIT clone and a copy and paste is the same really as this isn't a library or needs to be installed with other files from the same repo. But at least you're up and running - get learning, get tinkering, enjoy!