sqfmi/badgy #42: Cannot compile examples in Arduino IDE

sauerjc, 2019-08-13:
I just received my Rev 2B badgy.  I'm trying to compile and upload the new  bin file to my badgy using Arduino IDE 1.8.8  . I keep getting an ' stray error /'302' in program' when i compile the hello.ino example.  I have tried it with the original hello.h file and once that I created withe the image2cpp application.  I Even tried including all the  recommended libraries ensuring I used the specified version of the libraries.  I get the same error. 

the following is the text from the error message copied and pasted from the IDE

any help would be appreciated.  I thought this was going to be simple, cut and dry. 

TIA - john
-------------------------------------------------------------------------------------------

Arduino: 1.8.8 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

hello:31:3: error: stray '\302' in program

   <title>badgy/hello.ino at master Â· sqfmi/badgy Â· GitHub</title>

   ^

hello:31:3: error: stray '\267' in program

hello:31:3: error: stray '\302' in program

hello:31:3: error: stray '\267' in program

hello:268:5: error: missing terminating ' character

     </option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="115660478" data-scoped-search-url="/sqfmi/badgy/search" data-unscoped-search-url="/search" action="/sqfmi/badgy/search" accept-charset="UTF-8" method="get"><input name="utf8" type="hidden" value="&#x2713;" />

     ^

hello:314:7: error: stray '\342' in program

       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">â†µ</span>

       ^

hello:314:7: error: stray '\206' in program

hello:314:7: error: stray '\265' in program

hello:319:7: error: stray '\342' in program

       <span class="d-inline-block ml-1 v-align-middle">â†µ</span>

       ^

hello:319:7: error: stray '\206' in program

hello:319:7: error: stray '\265' in program

hello:355:7: error: stray '\342' in program

       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">â†µ</span>

       ^

hello:355:7: error: stray '\206' in program

hello:355:7: error: stray '\265' in program

hello:360:7: error: stray '\342' in program

       <span class="d-inline-block ml-1 v-align-middle">â†µ</span>

       ^

hello:360:7: error: stray '\206' in program

hello:360:7: error: stray '\265' in program

hello:387:7: error: stray '\342' in program
[... 406 lines cut ...]

hello:925:125: error: expected unqualified-id before '<' token

         <td id="LC35" class="blob-code blob-code-inner js-file-line">byte lastButtonState = <span class="pl-c1">0</span>;   <span class="pl-c"><span class="pl-c">//</span>the previous reading from the input pin</span></td>

                                                                                                                             ^

hello:929:186: error: expected unqualified-id before '<' token

         <td id="LC36" class="blob-code blob-code-inner js-file-line"><span class="pl-k">unsigned</span> <span class="pl-k">long</span> lastDebounceTime = <span class="pl-c1">0</span>;  <span class="pl-c"><span class="pl-c">//</span>the last time the output pin was toggled</span></td>

                                                                                                                                                                                          ^

hello:933:186: error: expected unqualified-id before '<' token

         <td id="LC37" class="blob-code blob-code-inner js-file-line"><span class="pl-k">unsigned</span> <span class="pl-k">long</span> debounceDelay = <span class="pl-c1">50</span>;    <span class="pl-c"><span class="pl-c">//</span>the debounce time</span></td>

                                                                                                                                                                                          ^

exit status 1
stray '\302' in program

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

chunkysteveo, 2019-08-19:
Looks like you copy and pasted the code from Github, and it's copied as HTML, not RAW C code?

sauerjc, 2019-09-04:
That's what I thought but I downloaded the examples rather than copying and pasting. I even tried "Tools>Fix encoding & Reload"  and "Tools/Autoformat"  in the Arduino IDE but still get the errors.

sauerjc, 2019-09-05:
all i am trying to do is get my badgy to display something like:

[image]

I'm getting no where fast when I cannot even compile the examples.  I'm running IDE 1.8.8 on Win10.  I do have access raspberrypi running linux and MacOSX if they help.

Thanks, John

sqfmi, 2019-09-20:
Hi @sauerjc , as @chunkysteveo mentioned, it looks like you are trying to compile the webpage's HTML code. Try downloading the code by clicking this button:
[image]
and try running the examples again. Let us know if you run into any other isssues.

closed 2019-09-20 by sqfmi