sqfmi/badgy #44: problem(s) with new Badgy

jdhysell, 2019-10-09:
I just received the new 2B model and started setting it up.
The first thing I see is that it has a bad solder joint on the mini-USB connector. I have to hold it at an angle to force it to connect to the circuit board... - very unpleasant for long sessions...
I was able to set it up to my local WIFI environment and proceeded to play with the examples.
I modified 'hello' to use my name, and verified and compiled it without issue.
But when I load it to the Badgy, with either WIFI or USB, it completes and informs me that Badgy is restarting. But nothing happens after that - the ink display remains with the connected for update image. Power cycling does nothing (other than letting me restart the OTA process if I want). I tried reloading the base version of 'hello' with the same end results.
Any suggestions?

sqfmi, 2019-10-15:
Hi @jdhysell , may we know the board settings used to compile the sketch?

jdhysell, 2019-10-15:
HiI read through the blog and found this setting page and followed it.
I wiped my initial Arduino 1.8.5 environment and restarted with 1.8.10
no change in how the badgy behaves.
The sketches all compile and behave like they transfer.A message appears that the unit will now reboot, and the wifi signalgoes away briefly. Then wifi returns *so it appears the badgy DOES reboot*but the e-paper never updates from the wifi update screen.
thanks for your supportJohn

jdhysell, 2019-10-15:
[image]

chunkysteveo, 2019-10-15:
Change Flash Size to 4M 1M SPIFFS, and try the upload again.

jdhysell, 2019-10-15:
sorryThe prior settings list was wrong
Flash is already set to 4M 1M SPIFFS
I am having trouble getting a snapshot/copy of the tools pulldown page...-John

jdhysell, 2019-10-15:
Is there any sensitivity to the version of the Adafruit GFX library used?I loaded 1.5.7
I know that version 2.x of  GxEPD is needed

sqfmi, 2019-10-15:
Hi @jdhysell, if it compiles and uploads then the version you have should be ok. Could you also check a few things:

1) Confirm you have Rev 2B and not Rev 2C
2) The ribbon cable is properly secured and not loose
3) Power cycling with the on/off button has any effect?

Thanks!

jdhysell, 2019-10-16:
It matches the 2B model on the Github home pageBut then I read the description of the 2C and it DOES appear to havea closed solder pad set marked * on the lower right corner.The protective film did have a green tabSo I am now thinking it is a 2CDo I need a special library for that?
I have reseated the e-paper ribbon cable several times - no joy there
Power switch does not impact the e-paper image at all.The only impact I can observe is the wifi signal dies with power in OFF position*as it should*
thanksJohn Hysell

chunkysteveo, 2019-10-16:
Can you connect to the Badgy via it's IP Address? 

When you power it up - is it creating a "Badgy" Wifi access point, or is it connecting to your home Wifi network (and you can ping it's IP address and connect to it via e.g. http://192.168.1.XXX:8888/update)?

Steve

jdhysell, 2019-10-16:
YesThat is how I have been uploading to it sincethe micro-USB connection has a bad solder joint.The upload code appears to work fine. After uploads, a messageindicates the unit will reboot. The IP address drops briefly *cannot ping it any more*.The IP address returns - but the e-paper does not change - sits on the code update screen.
thanks

jdhysell, 2019-10-16:
> @jdhysell with the closed solder * pad tab and a green tab, you should have Rev 2C and not 2B. With Rev 2C, you have to make sure you modify your code to use the GDEW029T5 driver instead of the GDEH029A1. You can do that by uncommenting line 4 and commenting out line 6 in hello.ino

OKI made those changes, recompiled, saved new bin and uploaded to the BadgyNo difference observed.The code uploaded and Badgy rebootedBut the e-paper remains on the /update page - power cycling does nothing.
In my copy of hello, the GDEH029A1 driver is referenced in line 3so it is still included after commenting line 4...
thxJohn

jdhysell, 2019-10-16:
I see that the example code was updated on 9/28I refreshed my copies and now can compile HELLO just fine
So the solution *for me* was to observe that the code has been updated, andthat I have a version 2C board.
thanks very much for your support.You can close this case.
regardsJohn Hysell

chunkysteveo, 2019-10-16:
//Use the GxGDEW029T5 class if you have Badgy Rev 2C. Make sure you are on GxEPD 3.05 or above

Are you using version 3 of the GxEPD? Your previous notes said you installed the V2 version...?

Steve

sqfmi, 2019-10-16:
Thanks @chunkysteveo for your support 👍 Glad to hear it's all working now @jdhysell

dnhuan, 2019-10-17:
sorry for the irrelevant question, did you order your rev2C on Tindie? Or if I buy Badgy on Tindie right now, will I get rev2C, or just 2B?

sqfmi, 2019-10-17:
@dnhuan if you ordered recently, it should be Rev 2C

jdhysell, 2019-10-17:
I received a 2C - I did not realize that until I was toldto examine the board more closely

sqfmi, 2019-11-18:
Resolved.

closed 2019-11-18 by sqfmi