sqfmi/badgy #45: Badgy no longer booting

bauerj, 2019-10-13:
Hey,

I just received my Badgy Rev 2C a few days ago. I tried to modify the "Hello" sketch but had issues uploading it onto the device. The connection was somewhat unreliable so I had to restart the process a few times until it worked. I noticed that the Badgy sometimes rapidly restarted itself (the display turned white and then the content appeared again) while the Arduino IDE tried to start flashing.

However after one failed attempt to flash a new program it seems like the Badgy is no longer booting at all. The screen doesn't change any more and I am unable to get into the web flash mode, which is still part of the sketch.

When turning the device on I see this:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c
```

in the serial monitor. No further output occurs.

Flashing a new sketch seems impossible, as the process stops with `esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header`.

I also tried using a different cable and a different computer but that did not solve the issue. Any idea what could be wrong here?

sqfmi, 2019-10-15:
Hi @bauerj , are you able to see the serial port show up in the Arduino IDE?

bauerj, 2019-10-15:
Yes, and when booting up the device I see the output posted earlier in the serial monitor.

sqfmi, 2019-10-15:
It appears to be crashing and stuck in a restart loop, and you are unable to upload a new sketch. Can you confirm if you are using a stable power source? Also, could you check and see if there are any loose connections or components (a picture of the back of Badgy would be really helpful if possible).

Thanks!

bauerj, 2019-10-17:
Not sure, how to check that. I used a different computer and the issue stayed the same. Measuring the voltage at the ESP-12F shows a stable 3.3V:

[image]

At the USB port I measure 5.03V but the pins are too tiny to take a photo while measuring.

Here is what the board looks like:

[image]

I can't see any loose connections with my naked eye.

bauerj, 2019-11-14:
Any idea how I could repair this?

sqfmi, 2019-11-14:
Hi @bauerj , try using the esptool to completely erase the flash and re-uploading the sketch, and make sure you are using the right compile settings. If all else fails, reach out to us on Tindie and we'll send you a new board.

liorbp, 2021-04-25:
@bauerj You could just pull down RST to low (means connecting a cable between RST and ground) and flash it again. It should work.
When you flash it make sure to do it with the _hello_ example (and don't forget to use the right GxEPD before doing so).

[image]