sqfmi/badgy #46: Not connecting via USB

laird, 2019-11-13:
I am unable to connect to Badgy on a MacBook Pro via USB. I was able to connect and program it with the Hello sketch from the repo via WiFi, once, after which it doesn't respond to WiFi, either. So I need USB programming to recover from that. I've tried several ports and several cables and installed some additional drivers suggested in various forums, but so far it's not recognized as a serial device on USB. Any suggestions?

sqfmi, 2019-11-14:
Hi @laird, do you have the drivers installed? You can find them here https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

Once installted, you should be able to see the device show up in the Arduino IDE e.g. /dev/cu.SLAB_USBtoUART

laird, 2019-11-15:
That did the trick. Thanks!

BTW, I didn’t see these drivers referenced in https://github.com/sqfmi/badgy <https://github.com/sqfmi/badgy> - you should add it. Or is there somewhere else that I should be looking for doc’s?

I’ve got sketches compiling, loading and running now, thanks!

I just noticed that the batteries I bought (Energizer 2450 Lithium battery) don’t seen ti be charging or powering Badgy. I am putting them in + up, which looks right. I did notice that the contact on the PCB (-) doesn’t stick up at all. Perhaps it’s not making contact?

sqfmi, 2019-11-15:
Awesome! We'll update this information in the README shortly.

closed 2019-11-15 by sqfmi