sqfmi/badgy #47: Compiler errors

mattpackwood, 2019-12-04:
I am getting errors and failure to compile, any suggestions?

Arduino: 1.8.10 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Multiple libraries were found for "DNSServer.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/libraries/DNSServer
Multiple libraries were found for "ESP8266WebServer.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/libraries/ESP8266WebServer
Multiple libraries were found for "ESP8266HTTPUpdateServer.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/libraries/ESP8266HTTPUpdateServer
Multiple libraries were found for "WiFiManager.h"
 Used: /Users/mattpackwood/Documents/Arduino/libraries/WiFiManager
Multiple libraries were found for "GxEPD.h"
 Used: /Users/mattpackwood/Documents/Arduino/libraries/GxEPD
Multiple libraries were found for "SPI.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/libraries/SPI
Multiple libraries were found for "Adafruit_GFX.h"
 Used: /Users/mattpackwood/Documents/Arduino/libraries/Adafruit_GFX_Library
Multiple libraries were found for "ESP8266WiFi.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/libraries/ESP8266WiFi
fork/exec /Users/mattpackwood/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3: no such file or directory
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JonPrevost, 2019-12-17:
I think the board should be NodeMCU 1.0 (ESP-12E Module), not Generic ESP8266 Module. I'm not sure though. I'm revisiting these boards and getting my own headaches compiling.

chunkysteveo, 2019-12-18:
Correct @JonPrevost [point 5](https://github.com/sqfmi/badgy#over-usb-for-rev-2a--rev-2b) states it needs to be NodeMCU 1.0 (ESP-12E Module) if compiling from Arduino IDE. Hope that helps @mattpackwood ?

mattpackwood, 2019-12-18:
That does not seem to help, sorry...

Multiple libraries were found for "WiFiManager.h"
 Used: /Users/mattpackwood/Documents/Arduino/libraries/WhareHauoraWiFiManager
Multiple libraries were found for "GxEPD.h"
 Used: /Users/mattpackwood/Documents/Arduino/libraries/GxEPD
Multiple libraries were found for "SPI.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/SPI
Multiple libraries were found for "Adafruit_GFX.h"
 Used: /Users/mattpackwood/Documents/Arduino/libraries/Adafruit_GFX_Library
Multiple libraries were found for "ESP8266WiFi.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi
Multiple libraries were found for "DNSServer.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/DNSServer
Multiple libraries were found for "ESP8266WebServer.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WebServer
Multiple libraries were found for "ESP8266HTTPUpdateServer.h"
 Used: /Users/mattpackwood/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266HTTPUpdateServer
fork/exec /Users/mattpackwood/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3: no such file or directory
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

mattpackwood, 2020-01-01:
It looks like the badgy is going to recycling...  Oh well...

chunkysteveo, 2020-01-01:
There's no issue with the Badgy board. Those errors are errors in your Arduino IDE compiling your files. I would go back to basics and make sure you can compile and upload a basic blink sketch to the board,  and then some ESP8266 example sketches. It's a great little board and worth getting working. Help @sqfmi ?

Steve

JonPrevost, 2020-01-01:
I'm sure your troubles are in the IDE. I had to muck around with different versions of libraries to get even the example codes to compile. Still not entirely sure of the multiple libraries warnings. I can uninstall and it's still there warning me of multiple versions of spi.h etc.. I noticed that Json wasn't rolling back to an older version so that was a bug I ran into. Good luck with the IDE help, I dont even know how I got it working. The board works really well but it looks less than professional. When adding in the price this exceeds expectations for a nice development board for the esp module.

lasniko, 2020-01-10:
It clearly states the problem:

> exec /Users/mattpackwood/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3: no such file or directory

check your library structure

mattpackwood, 2020-01-10:
Hi @lasniko,
There is a file there...

lrwxr-xr-x  1 mattpackwood  staff  22 Dec 17 17:03 /Users/mattpackwood/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -> /usr/local/bin/python3

And the target is there, am I missing something?

mattpackwood, 2022-04-30:
Closed as no longer needed

closed 2022-04-30 by mattpackwood