sqfmi/badgy #48: [micropython] update example for the latest display

dadokkio, 2019-12-04:
Hi, not sure why there is no "new issues" on badgy-python so I'll open the issue here.

Is "epaper2in9.py" working with the latest revision of badgy? 
There is an updated "greyscale" demo available also for micropython?

Thanks.

sqfmi, 2020-01-05:
Updated in https://github.com/sqfmi/badgy-python

closed 2020-01-05 by sqfmi

dadokkio, 2020-01-07:
I tried to print an image following the arduino grayscale tutorial for creating the image array.
Not sure if the setting are correct (16 color?) but the image seems to be huge.
Secondly, if I need to write light gray or dark gray text there is an example on how to do that?

Thanks

dadokkio, 2020-01-08:
I made also other test plotting lines and text with framebuffer but the only way to render them correctly is with framebuf.MONO_HLSB with badgy in portrait mode.