sqfmi/badgy #49: not able to upload  firmware over wifi - OTA

terrence002, 2020-01-02:
Rev 2c was able connect, to wifi network, I can see  ESP_AC9D8A as a connected device in the router connected list.
however badgy seems to be going in loop with it blinking from "Hello badgy" and a blank screen. 
http://YOUR_IP_ADDRESS:8888/update  command does not do anything.

chunkysteveo, 2020-01-02:
Just to confirm - you're not typing in the URL "YOUR_IP_ADDRESS", but the IP of the Badgy in your router connected device list, e.g. 192.168.1.123 or 192.168.0.123 etc? Also to get to the update, you need to hold down the joystick button on boot up.

But.... if it's flashing between the hello graphic and blank, it looks like it's stuck in a boot loop. Does it work connected to USB power, not battery?

terrence002, 2020-01-02:
That is correct i am using the ip instead for "YOUR_IP_ADDRESS",
I am using a USB power for the badgy, not battery.

sqfmi, 2020-01-02:
Was it in a boot loop before connecting to WiFi? Also, are you able to compile the Hello sketch and upload it to Badgy?

terrence002, 2020-01-11:
not sure but it looks like the screen is stuck at ipaddress:8888/update screen, not able to compile the hello sketch and upload to Badgy.  and the ip address url does not work