sqfmi/badgy #50: Console log not showing anything in Serial (COM port)

Dacesilian, 2020-01-02:
I'm trying to debug my program (badgy connected with USB), so I'm using this in `setup()` method:
```
Serial.begin(115200);
Serial.println("Setup");
display.init(115200);
```
But nothing is showing in terminal (only unreadable characters on boot).

Is Serial supported? Is speed 115200 okay? I've tried 9600 as well, but the same result.
When I open Tools -> Serial monitor (in Arduino IDE), badgy does not restart (as other devices do).

Thanks.

sqfmi, 2020-01-02:
If you call display.init with a baud rate, it will try to initialize the Serial port again. Try just calling display.init() without a parameter. Serial is supported and 115200 should be fine.

Dacesilian, 2020-01-03:
Yes, I've excluded Serial.begin, as it is defined also in display.init(115200), but no text in console.

[image]

[image]

[image]

[image]

There was a readable text in console when connecting to WiFi, but my Serial.println is beeing ignored.

Dacesilian, 2020-01-03:
I've made a few tests and my observation is that console is fine with this code:

```
#include <GxEPD.h>
#include <GxGDEW029T5/GxGDEW029T5.h>
#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>

GxIO_Class io(SPI, SS, 0, 2);
GxEPD_Class display(io); // default selection of D4, D2

void setup() {
  display.init(115200); // enable diagnostic output on Serial
  display.setRotation(3);
  Serial.println("Setup");
  Serial.print("..... :) .....");
  Serial.println();

  /*pinMode(1, INPUT_PULLUP); //down
    pinMode(3, INPUT_PULLUP); //left
    pinMode(5, INPUT_PULLUP); //center
    pinMode(12, INPUT_PULLUP); //right
    pinMode(10, INPUT_PULLUP); //up*/

  Serial.println("Setup done, yeah");
}
```
[image]

But when I enable pinMode lines, it won't show anything (even not timestamp in console):
[image]

chunkysteveo, 2020-01-03:
Ahh yes. The button is using the TX/RX pins. When you enable them, they are inputs, so no serial protocol output. I am sure I saw in some readme that you can't use serial output if using buttons - a vague memory I think??!

Steve

chunkysteveo, 2020-01-03:
The shared pins with TX/RX for the buttons is a limitation of the ESP8266 module and the fact the paper display needs so many pins to operate. It's either use serial output, or use buttons, not both.

chunkysteveo, 2020-01-03:
Found it - [https://github.com/sqfmi/badgy/issues/27#issuecomment-434373427](https://github.com/sqfmi/badgy/issues/27#issuecomment-434373427)

Dacesilian, 2020-01-03:
Thanks, it's clear now. It's sad.

closed 2020-01-03 by Dacesilian