sqfmi/badgy #58: Esphome support?

jlongman, 2022-11-03:
Is badgy supported by esphome? Does anyone have config files if so?

@chunkysteveo you tagged a post with `esphome` [here](https://www.instagram.com/p/CgJeIZorzWx/)

The waveshare eink displays seem similar and closest, but different enough it seemed worth asking. 

(Rev 2A if it matters)

chunkysteveo, 2022-11-04:
It is "supported" by ESPhome, yes.... depending on what you want to do with it...?!

You can use the ePaper screen (with partial refresh), and the 5-way switch. Accessing the onboard (ESP8266) as a status light seemed to cause issues, so I don't include that.

The ADC can also be included to measure the voltage level of the battery. But be aware the voltage divider used isn't right for a lipo battery.

There's a lot of pins used on the ESP8266 which tend to interfere with the OTA process by some line going low (I think?!), so I also include the "Safe mode reboot" switch, which seems allows for 100% guarantee upload (weeks of troubleshooting and testing!).

Do you have a specific example you'd like to set the Badgy up with, I can post an example YAML config?

chunkysteveo, 2022-11-04:
[image]
[image]

Here's two examples I have using Badgy and Home Assistant using ESPHome. The buttons aren't really used, but they could be to move between pages of content as an example.

chunkysteveo, 2022-11-04:
> @chunkysteveo you tagged a post with `esphome` [here](https://www.instagram.com/p/CgJeIZorzWx/)

Yeah, that example was also another version I had - before I updated it to the hourly/daily forecast one seen above. Those "weather" examples use a custom sensor within Home Assistant to format a bunch of other sensors into a simple text sensor that the Badgy can process and display.

jlongman, 2022-11-04:
Amazing! I’m interested in both of those, yes, and I also have a co2 monitor that I might do on its own in as- large-as-possible text. Getting the switches going might be interesting to flip between them, later, but as I have two of them I can start without it.

chunkysteveo, 2022-11-06:
Let me put a 'hello world' config yaml file up here for you to start on, which should give you some good pointers and a good base... how OK are you with ESPHome and HA?

jlongman, 2022-11-06:
A ‘hello world’ would help a lot, thanks!

Okay with esphome, I’ve put some basic sonoff switches, a co2 meter together; and competent with HA, I’ve wrestled and won with templates, etc.

chunkysteveo, 2022-11-07:
Not had any time to gather all my scripts and info together, but below is a quick and dirty "hello world" which should get you working as a proof of concept:

In HA, you will need weather.home (Met.no weather service), binary_sensor.door_sensor_back_door_contact (a binary sensor such as a door sensor),  sensor.multisensor_garage_temperature (a temperature sensor of some sort), binary_sensor.multisensor_garage_motion (a motion sensor binary sensor such as a PIR sensor. This is to blank the screen if no one is present to save on the epaper life, comment out the if else statement if you want it to always be on), and you will need to download and add some fonts to the ESPHome directory (fonts/materialdesignicons-webfont.ttf, fonts/materialdesignicons-webfont-6.9.96.ttf (I need to combine these two MDI calls into one (lazy!), feel free to fix!), fonts/Roboto-Bold.ttf, fonts/Roboto-Regular.ttf)

Hopefully the below will give you some ideas and paths to explore. The end result will be a vertical Badgy with two temperature readings, a weather cloud/sun icon from Met.no service, and a CLOSED/OPEN and door icon at the bottom, along with the update time, battery icon, and wifi range icon. This is a mis-match of a bunch of versions I have running, so it's cut back from what I run, but has a lot of things in there that I have learned..... Use the Safe Restart button if you are having issues OTA updates to Badgy.

```
substitutions:
  display_name: 'EPaper Display Badgy 01'
  device_description: 'EPaper Display Badgy 01. The White one!'
  friendly_name: 'EPaper Display Badgy 01'
  device_name: epaper-display-01
  node_name: epaper_display_01
  
globals:
  - id: data_updated
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: initial_data_received
    type: bool
    restore_value: no
    initial_value: 'false'
    
esphome:
  name: ${device_name}
  comment: ${device_description}
  on_boot:
    - priority: 200.0
      then:
        - component.update: epaper
    - priority: 800.0
      then:
        - component.update: ${node_name}_uptime_raw
        - component.update: wifisignal
      
esp8266:
  board: nodemcuv2
  
# Enable logging
logger:
  # level: debug

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password
  num_attempts: 5

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Epaper-Display-01"
    password: "PdfmXwsyIhdgEjsxnf"

captive_portal:

time:
  - platform: homeassistant
    id: homeassistant_time
  
#light:
#  - platform: status_led
#    name: ${display_name} Onboard LED
#    pin:
#      number: GPIO2
[... 414 lines cut ...]
            it.printf(28+100, y_top+32, id(temp_font), TextAlign::BOTTOM_RIGHT , "%5.1f°C", id(temp_garage).state);
          }
          // Print Solar Weather Station temperature (from homeassistant sensor)
          if (id(temp_solar_weather).has_state()) {
            it.printf(34, y_top+32+32-3, id(weather_font), TextAlign::BOTTOM_RIGHT, "%s", weather_icon_map[id(weather_forecast_today).state.c_str()].c_str());
            //it.print(34, y_top+32+32-3, id(icon_font_new), TextAlign::BOTTOM_RIGHT, "\U000F18D6");
            it.printf(28+100, y_top+32+32, id(temp_font), TextAlign::BOTTOM_RIGHT , "%5.1f°C", id(temp_solar_weather).state);
          }
        
          
          //DOOR
          if (id(door_backdoor).has_state()) {
            it.printf(64, 220, id(temp_font), TextAlign::TOP_CENTER, "%s", id(door_backdoor).state ? "OPEN" : "CLOSED");
            if (id(door_backdoor).state) {
              it.print(2, 250, id(icon_font_new), "\U000F081C");
            } else {
              it.print(2, 250, id(icon_font_new), "\U000F081B");
            }
          }          
        }
        else{
          it.fill(COLOR_OFF);
        
        }
```

jlongman, 2022-11-12:
Finally some time to get back to this, and preliminary results are good!  

First attempts I was seeing nothing but either I watching the logs wirelessly prevents it from displaying or maybe I had `update_interval` too low or wasn't patient enough.

Now that I've got the basics I'll look at getting my CO2 levels in, thanks muchly!

chunkysteveo, 2022-11-13:
Ahhh - i wondered when you'd look at this. Good to hear it's working. I have similar results with blank screens, reboots etc. Not sure if it's the device wiring or the code, but it's not the most stable ePaper screen for HA. 

Enjoy tinkering with the YAML! Check out https://github.com/Madelena/esphome-weatherman-dashboard for some good inspiration too. And there are a bunch of epaper esphome projects to borrow ideas from!

jlongman, 2023-12-22:
BTW, I'd given up on it for a while and checked in again and its working brilliantly now!  I think something in esphome improved more than any change I did.  Now I have to find the other one I've misplaced!

Showing:
- time
- indoor temp
- outdoor temp
- co2
- status of front and back doors
- back door on right, front door on left
- last update time and wifi strength.

[image]

chunkysteveo, 2023-12-22:
> BTW, I'd given up on it for a while and checked in again and its working brilliantly now! I think something in esphome improved more than any change I did. Now I have to find the other one I've misplaced!
> 
> Showing:
> 
> * time
> * indoor temp
> * outdoor temp
> * co2
> * status of front and back doors
> * back door on right, front door on left
> * last update time and wifi strength.
> 
> ![badgy](https://private-user-images.githubusercontent.com/1051995/292525438-15540b4b-6b42-44d6-993b-0d6bf63ac6c0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMyODY3NTUsIm5iZiI6MTcwMzI4NjQ1NSwicGF0aCI6Ii8xMDUxOTk1LzI5MjUyNTQzOC0xNTU0MGI0Yi02YjQyLTQ0ZDYtOTkzYi0wZDZiZjYzYWM2YzAuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjJUMjMwNzM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjViMzg0MDkwOWU0ZDQ0YmM2NTg4NTVjYzZiOWM3ZTYzOThlZTg1Njg1Y2FlZTI4MzczZmI1ZTJjN2M5MWM0NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Kzb8fiODs4cmrYYX2qM2RaXrOQsCGspSqLq49LkmQ5g)

Haha! Looking awesome!!! Glad it is working for you. Mine has lived on my wall ever since with all sorts of temps and door icons!